zephyrproject-rtos/west #257: Git LFS support for remote not named "origin"

bengartner, 2019-05-03:
If you have a project with git LFS, and the remote name in your manifest does not happen to be origin, git lfs will give a cryptic error, and leave the repo in a bad state.

```
running 'git checkout --detach refs/heads/manifest-rev' 
Downloading LargeFile.csv (1.0 MB)
Error downloading object: LargeFile.csv (241fb7f): Smudge error: Error downloading LargeFile.csv (241fb7f406f53b1167319c99e7f6329d028158c30c868bc619a22548ac61b0ec): batch request: missing protocol: ""
```

This is largely a git lfs issue, and they have a number of proposed and/or implemented solutions. 
https://github.com/git-lfs/git-lfs/issues/2710

However, I think the minimal state that west puts the git repo is causing some problems with git lfs.

The basic issue is that git lfs waits to pull down large file objects until a checkout, at which point it has to work backwards and try to determine which remote has the large file objects for the commit in question. The project I've been using only has one west remote, so it should be able to find it, but west could probably also set up a tracking or default branch, or explicitly fetch lfs objects as a workaround or to improve robustness.

mbolivar, 2019-05-06:
Hi, thanks for the report!

Git creates the remote named "origin" but west never uses remotes by name. It actually only uses URLs to fetch. If you enable the most verbose output (`west -vvv update`) you can see this in the git commands it logs:

```
running 'git fetch -f --tags -- https://github.com/zephyrproject-rtos/tinycbor 'refs/heads/*:refs/west/*'' in C:\Users\Marti\zephyr\modules\lib\tinycbor
````

So I'm not sure this is about the remote name. Can you provide an example for me to repro or some more details on your setup?

Thanks!

bengartner, 2019-05-06:
Sure! I created a minimal example of the problem. Let me know if you see the same error I quote below.

https://github.com/bengartner/examplewestproj

west never uses remotes by name directly, but git lfs does. As far as I understand, I'm not an lfs expert. But my understanding is git lfs waits until you actually checkout files before fetching any files marked "large," so when west runs a 'git checkout' there is a hook that git lfs uses to go fetch any large files - at which point it has to determine where to fetch them from.

```
west init -m https://github.com/bengartner/examplewestproj
west -vvv update
```

The update fails with:
```
running 'git checkout --detach refs/heads/manifest-rev' in C:\Users\bengartner\source\tmp52\examplewestproj\dataset
Downloading dataset.csv (16 MB)
Error downloading object: dataset.csv (843ca39): Smudge error: Error downloading dataset.csv (843ca39fb344d8415ff51620b67addd87e06887e1080947e7c8779919ecd9820): batch request: missing protocol: ""

Errors logged to C:\Users\bengartner\source\tmp52\examplewestproj\dataset\.git\lfs\logs\20190506T095903.2231696.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: dataset.csv: smudge filter lfs failed
'git checkout --detach refs/heads/manifest-rev' in C:\Users\bengartner\source\tmp52\examplewestproj\dataset finished with exit status 128
FATAL ERROR: Command 'checkout --detach refs/heads/manifest-rev' failed with code 128 for examplewithlfs (examplewestproj\dataset)
```

mbolivar, 2019-05-25:
Thanks @bengartner -- just letting you know that I haven't forgotten about your report and will work on fixing this.

I've never used LFS, but I think supporting it is important enough to fix this issue. In the Zephyr case, I see LFS as an interesting possibility for letting west pull down toolchains on platforms where the Zephyr SDK is not available. The most important one is of course Windows, which most of my company's customers use for embedded development :).

bengartner, 2019-05-26:
@mbolivar Great, thanks. I've found a workaround that works for use, but I can help check that the Windows setup works. When you have time, if you're able to reproduce, let me know if you think a workaround in west is appropriate or if I should file as a git LFS bug.

mbolivar-nordic, 2021-07-24:
This may be fixed by https://github.com/zephyrproject-rtos/west/pull/514.