zephyrproject-rtos/west #258: Problem with `west init` in Docker container

acihy, 2019-05-07:
I have problems with the `west` in a Docker container. I've mounted a folder with [zephyr](https://github.com/zephyrproject-rtos/zephyr) and I had to use `west init -l <dir-to-zephyr-folder>` after every Docker container launch. It worked but after [this commit](https://github.com/zephyrproject-rtos/west/commit/cb660b6e28ce96c956e19a1244dac1df938e5ba3) when I want to built an example program I receive the following error:

```
build@45aafadd238d:~/zephyr/samples/basic/blinky/build$ cmake -DBOARD=nrf52_pca10040 ..
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Selected BOARD nrf52_pca10040
Traceback (most recent call last):
  File "/usr/lib/python3.6/configparser.py", line 1138, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'manifest'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/build/.west/west/src/west/manifest.py", line 60, in manifest_path
    config.get('manifest', 'path'),
  File "/usr/lib/python3.6/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.6/configparser.py", line 1141, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'manifest'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/west", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/west/_bootstrap/main.py", line 499, in main
    wrap(wrap_argv)
  File "/usr/local/lib/python3.6/dist-packages/west/_bootstrap/main.py", line 485, in wrap
    west.main.main(argv)
  File "/home/build/.west/west/src/west/main.py", line 556, in main
    args.handler(args, unknown)
  File "/home/build/.west/west/src/west/main.py", line 290, in command_handler
    command.run(known_args, unknown_args, topdir)
  File "/home/build/.west/west/src/west/commands/command.py", line 111, in run
    self.do_run(args, unknown)
  File "/home/build/.west/west/src/west/commands/project.py", line 287, in do_run
    for project in _projects(args):
  File "/home/build/.west/west/src/west/commands/project.py", line 620, in _projects
    projects = _all_projects()
  File "/home/build/.west/west/src/west/commands/project.py", line 689, in _all_projects
    return list(Manifest.from_file().projects)
  File "/home/build/.west/west/src/west/manifest.py", line 85, in from_file
    source_file = manifest_path()
  File "/home/build/.west/west/src/west/manifest.py", line 64, in manifest_path
    .format(e.args[0])) from e
west.manifest.MalformedConfig: missing key: 'manifest' in west config file
-- Loading /home/build/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040.dts as base
-- Overlaying /home/build/zephyr/dts/common/common.dts

warning: UART_CONSOLE (defined at drivers/console/Kconfig:46) was assigned the value 'y' but got the
value 'n'. You can check symbol information (including dependencies) in the 'menuconfig' interface
(see the Application Development Primer section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.

warning: the choice symbol UART_0_NRF_UART (defined at drivers/serial/Kconfig.nrfx:27) was selected
(set =y), but no symbol ended up as the choice selection.  You can check symbol information
(including dependencies) in the 'menuconfig' interface (see the Application Development Primer
section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_0_NRF_UART.html (which is updated
regularly from the master branch). See the 'Setting configuration vaParsing Kconfig tree in /home/build/zephyr/Kconfig
Loading /home/build/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040_defconfig as base
Merging /home/build/zephyr/samples/basic/blinky/prj.conf
Configuration written to '/home/build/zephyr/samples/basic/blinky/build/zephyr/.config'
lues' section of the Board
Porting Guide as well.
-- Cache files will be written to: /home/build/.cache/zephyr
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/build/zephyr/samples/basic/blinky/build
```

I tried to use the following versions: master, v0.5.6, v0.5.7, v0.5.8, but it didn't help. 
Does anybody have an idea how to fix it?

`west -V` output:

```
West bootstrapper version: v0.5.8 (/usr/local/lib/python3.6/dist-packages/west/_bootstrap)
West repository version: v0.5.8-7-gcb660b6 (/home/build/.west/west)
```

mbolivar, 2019-05-10:
What version of zephyr are you using? I couldn't reproduce.

acihy, 2019-05-13:
I'm using `v1.14rc1`.

mbolivar, 2019-05-13:
> I'm using v1.14.0rc1.

Yeah, that won't work anymore because the rc1 manifest set the west revision to `master`, which has since had incompatible changes merged. Your options are:

1. update to a more recent version (`v1.14.0`, e.g.)
2. set the `revision` field in the `west` section to `v0.5.8` in your tree

closed 2019-05-13 by mbolivar