zephyrproject-rtos/west #259: Support FreeBSD

sdalu, 2019-05-07:
FreeBSD: 
* system-wide:   `/usr/local/etc/westconfig`
* user-wide: `~/.westconfig` or `$XDG_CONFIG_HOME/west/config`

mbolivar, 2019-05-09:
What happens now when you run on freebsd? Is there some error?

sdalu, 2019-05-09:
The following error is due to a missing test in `ConfigFile` (file `configuration.py`), where `paltform.system()` is not tested for FreeBSD

I'm guessing also that `read_config` should be modified as an explicit test is made on `platform.system()`

~~~
Traceback (most recent call last):
  File "/home/sdalu/.local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/home/sdalu/.local/lib/python3.6/site-packages/west/_bootstrap/main.py", line 499, in main
    wrap(wrap_argv)
  File "/home/sdalu/.local/lib/python3.6/site-packages/west/_bootstrap/main.py", line 485, in wrap
    west.main.main(argv)
  File "/home/sdalu/zephyrproject/.west/west/src/west/main.py", line 538, in main
    config.read_config()
  File "/home/sdalu/zephyrproject/.west/west/src/west/configuration.py", line 95, in read_config
    files.append(ConfigFile.SYSTEM.value)
  File "/usr/local/lib/python3.6/enum.py", line 326, in __getattr__
    raise AttributeError(name) from None
AttributeError: SYSTEM
~~~

mbolivar, 2019-05-25:
Hi @sdalu -- if you have a chance to try out #271 and see if it works for you, I'd appreciate it.

To test, run `pip install -e .` from the west repository with that PR checked out, then try out the use cases that matter to you. If you're happy, I'll mark that PR as fixing this issue.

Thanks again for your report.

sdalu, 2019-10-04:
Just had time to try zephyr again. And it is now possible to install west on FreeBSD.