zephyrproject-rtos/west #262: Offline usage

MrPointer, 2019-05-14:
Hello all,

My team and I are developing embedded apps using the Zephyr RTOS for a while now, and we've recently decided to upgrade to its' 1.14 version. However, we were surprised to see a new policy/behavior mostly regarding board flashing, as it's now using the **west** tool by default.
That's welcome of course, as west should ease the lives of devs eventually, but for offline devs as we are - It really complicates things.
This is mostly because the `west init` command attempts to fetch and clone some "_stuff_" from the internet, which we don't have on our workstations - at all.
From my understanding, this "_stuff_" is repositories required to successfully develop in the Zephyr environment, such as the Zephyr source itself, and this is all driven by the manifest file, which can be configured otherwise, including listing custom URLs which point to the relevant repos.

Given all of this - What should be the recommended way to use the tool in offline environments?

We thought about "mirroring" all repos required by the default manifest from an online machine to an inner-org git server, then creating a custom manifest file as described above, where the URLs will point to our inner-org server instead.
Nevertheless, we want to ask you, maintainers - How'd you do it?

mbolivar, 2019-05-15:
Hi -- `west init` and `west update` are separate commands from `west flash`. If you're just building and flashing, there is no need to have an internet connection. So I don't think there is an issue here, but I may be missing something.

If you run `west init` or `west update`, you do indeed need a network connection, same as if you ran `git clone` or `git fetch`, but the flash/debug targets don't do that.

Does that help? Please close the issue if so, or if I'm not understanding you properly please let me know what I missed.

Thanks!

mbolivar, 2019-05-23:
I'm going to close this as I think I've answered your question. Please feel free to reopen.

MrPointer, 2019-05-25:
Hi, sorry for the inactivity, been busy doing some other things but also verifying our issue.
Well, indeed you've misunderstood me 😄 

Let me try to describe our scenario with a step-by-step description:
1. Download the latest _stable_ zephyr version from GitHub, which would download the source code from a connected machine, i.e. one that has internet access
2. Download the latest SDK as well
2. Move downloads to an offline environment
3. Place it somewhere in a user's file system, extracting and setting up environment to use Zephyr as required
4. Run **CMake** on a project which uses Zephyr - Could be a sample from Zephyr's source itself
5. Build the project
6. Try to flash by running e.g. `ninja flash`

It would fail with an error message about missing the **west** tool, which is correct since it hasn't been installed yet (Following the steps above). So we installed it from a downloaded pip resource (such as **wheel**), and tried to flash again, but the problems persist as now it doesn't have the required sources for it to run as it should.

After attempting the proposed solution I've written earlier in the issue's 1st comment, we've managed somehow to configure it to work "as expected", and perform the flash.
However, I still wouldn't consider it a "Success" as it's unclear whether we've done it the right way or not, and whether it has be to that tough to perform a simple offline operation such as `flash`.

P.S
Sorry for not attaching any output or exact error messages, not near the workstation right now, and besides, it's really hard to copy by-hand from an offline machine...

MrPointer, 2019-05-29:
@mbolivar Can this issue be reopened please?

mbolivar, 2019-06-04:
Hi @MrPointer, sure:

> It would fail with an error message about missing the west tool, which is correct since it hasn't been installed yet (Following the steps above). So we installed it from a downloaded pip resource [...]

I'm not sure what the problem here is. If you install west at the same time as all of Zephyr's other dependencies in your step 2, then there should be no problem here. Nothing after step 2 requires a network connection, assuming step 3 is just a wholesale copy of the entire zephyrproject directory, including .west and all the modules.

> Sorry for not attaching any output or exact error messages, not near the workstation right now, and besides, it's really hard to copy by-hand from an offline machine...

If by "offline" you mean "airgapped" (i.e. never connected to the network), any chance you can save the messages to a file and then move that file to a networked machine?

mbolivar, 2019-06-04:
TL;DR zephyr requires a bunch of other python stuff to build, like pyelftools. If those are available in your offline environment then west should be able to be put there too. West flash does *not* require a network connection.

carlescufi, 2019-06-05:
@MrPointer as @mbolivar described, `west` is a Python tool that needs to be installed alongside any other Python requirements, so if want to follow a very strict offline policy and don't want to mix installing Python tools in multiple places you could do (pseudo code):

```
<install Git>
<install toolchain, CMake, etc..>
mkdir zephyrproject
cd zephyrproject
git clone git@github.com:zephyrproject-rtos/zephyr.git
pip3 install --user -r zephyr/scripts/requirements.txt
pip3 install --user west
west init -l zephyr/
west update
```
After that you are all set, `west` doesn't need any internet connectivity at all to build, flash or debug.

MrPointer, 2019-06-05:
@mbolivar @carlescufi I never thought or meant that `flash` requires internet connection, I used it merely to show an example of a command which I assumed was **_not_** relying on `west`.
With that said, the installation of `west` itself does require internet connection to:
1.  To get the zephyr source, but that can be avoided by transferring one from an online machine and using the `-l` option as stated by @carlescufi.
2. To download additionally required modules as stated in a **manifest file**, but that can't really happen on an "airgapped" machine, can it? **UNLESS** - Those modules would be somehow stored in a private, offline 
git repository, and zephyr's `west.yml` would be modified to point remotes to this repository.
But that's quite cumbersome to do, since every update to zephyr itself would require to change that `west.yml` file yet again.

All I'm saying is there doesn't seem to be an obvious way to correctly setup the dev environment on offline machines. While applying my suggested solution with private git repos might work for some, others do not even have such repositories, and may depend on a local-only solution - That's currently simply not provided (or not clearly enough).
BTW, if @carlescufi's meaning was to do all the steps above on an online machine, then transfer it to an offline one, it might actually work, but then again - I'm not sure if that's the most convenient one.

I guess I'm searching for explicit answers from you guys - a decision, which will be considered the standard and accepted way to do this.

carlescufi, 2019-06-05:
@MrPointer 

>BTW, if @carlescufi's meaning was to do all the steps above on an online machine, then transfer it to an offline one, it might actually work, but then again - I'm not sure if that's the most convenient one.

Yes, that is what I was suggesting. 
Imagine we were using only standard Git and no west at all. You would still need to clone the repository from somewhere, meaning you'd probably end up either cloning with an online machine, then hosting that cloned repo somewhere in your network, then pulling that into your airgapped machines in some way. All that `west update` does is pull down a bunch of Git repos automatically, which are all then stored inside the enclosing folder (`zephyrproject` in my example).
So one solution you have is to run what I posted above on the online machine and then copy the contents of the `zephyrproject` folder (all of it, including the hidden `.west` folder) into your airgapped machines. This should work since the repos would be preserved as-is, and we only use relative paths to the enclosing folder.

That said your usecase is very interesting and I'd certainly be glad to try to accommodate to the extent possible. @mbolivar has perhaps additional feedback.

mbolivar, 2019-06-05:
> @mbolivar @carlescufi I never thought or meant that `flash` requires
> internet connection, I used it merely to show an example of a command
> which I assumed was **_not_** relying on `west`.  With that said, the
> installation of `west` itself does require internet connection to:
> 1.  Get the zephyr source, but that can be avoided by transferring one
>     from an online machine and using the `-l` option as stated by
>     @carlescufi.
> 2. Download additionally required modules as stated in a **manifest
> file**, but that can't really happen on an "airgapped" machine, can
> it? **UNLESS** - Those modules would be somehow stored in a private,
> offline git repository, and zephyr's `west.yml` would be modified to
> point remotes to this repository.  But that's quite cumbersome to do,
> since every update to zephyr itself would require to change that
> `west.yml` file yet again.

OK, so I think the goal here is to create a complete west installation,
then take the machine "offline"?

But I'm still confused about what your use case is; sorry:

> Given all of this - What should be the recommended way to use the tool
> in offline environments?

This is from the original issue. I usually interpret "offline" to mean
"no network connection". The term "airgapped" is an even stronger one
that's usually used to refer to a machine which has never and will
never be taken online, e.g. machines with no network interface cards (NICs).

> We thought about "mirroring" all repos required by the default
> manifest from an online machine to an inner-org git server, then
> creating a custom manifest file as described above, where the URLs
> will point to our inner-org server instead.

... but looking again, it says here in the original question that
there is a possibility to contact an "inner-org" server (does that
mean a server on your LAN?).

So I don't think the environment is really either "offline" or
"airgapped", after all.

> Nevertheless, we want to ask you, maintainers - How'd you do it?

It really depends on the details of your environment and what you're
trying to do.

What kind of network connectivity do you have?

1. None at all, not ever, not even a LAN, this machine never has any
   NIC hardware (airgapped)
2. None after a certain point in time, though the machine
   does have a NIC which can connect to the Internet during a "setup"
   phase and is then removed or disabled (offline)
3. None after a certain point in time, but machine has a NIC which can
   connect to the Internet during a setup phase and a LAN only
   afterwards (I'll call this "offline with LAN")
4. Though the machine has a NIC which can connect to a LAN at
   arbitrary points in time (I guess I'll call this "LAN-only")

What are you trying to do with zephyr?

- Just develop zephyr as separate individuals?
- Share results with others in your team on the LAN?

Rather than break down exactly what I would do in each scenario
(that's already 8 cases), I'll name the requirements using the terms
in these docs:

https://docs.zephyrproject.org/latest/guides/west/repo-tool.html

If you want to have a fully functional zephyr installation, you need:

1. west bootstrapper installed, using pip
2. zephyr as the manifest repository cloned into an installation,
   using west init (not "west init -l"; don't use that unless you're
   moving a standalone, pre-v1.14 zephyr repository to a multi-repo
   west based installation on an online computer; that's what -l is
   for, and not much else)
3. west itself and other zephyr modules cloned, using west update

If you have all that, you can develop Zephyr 100% without any more use
of a NIC.

You could even copy that directory structure to an airgapped machine,
e.g. on a USB stick, and develop there.

If you want to share results with others, you have *some* network
connectivity, even if it's just a LAN. If you don't have WAN
connectivity, then I'd fork the manifest repository (i.e. zephyr) and
change the default remote to point at a server you can access on the
LAN, and use that to host your Git repositories.

Hope that helps.

MrPointer, 2019-06-07:
@mbolivar @carlescufi Thank you both for your thorough answers, it covers pretty much all possible scenarios.
In fact, it does seem I've mislead you to think that my machine is air-gapped, although it **does** have a LAN connection to our org's internal network, which also happens to host a private git server.
So for me, the solution suggesting to recreate the required repositories (zephyr source, west modules, etc.) on our inner-org's git server suits best, as it allows fellow developers to access it as well.

Nevertheless, I think all of the proposed solutions should be part of the documentation. I believe that many devs will find it valuable. Currently I'm out-of-time, but I'll be happy to contribute once I'll have the opportunity!

Thank you for your help!

mbolivar, 2019-06-11:
> So for me, the solution suggesting to recreate the required repositories (zephyr source, west modules, etc.) on our inner-org's git server suits best, as it allows fellow developers to access it as [...]

OK, great! Happy to review any docs patches along these lines. Thanks!

closed 2019-06-11 by mbolivar