zephyrproject-rtos/west #263: west's build/run support is not adequate for real usage

pfalcon, 2019-05-14:
Let's do `west build --help` and read something up:

~~~
  -b BOARD, --board BOARD
                        Board to build for
  -d BUILD_DIR, --build-dir BUILD_DIR
                        Build directory. If missing and run in a Zephyr build
                        directory, it is used; otherwise, it's "build". Always
                        created if it doesn't exist.
~~~

What it says there is that regardless of BOARD specified, it will always use one, the same build dir, or another words, at the same time, building for only one board is supported! It's actually frightening to think that west promotes such a workflow. I can't call myself too advanced a Zephyr developer, but I all the time build for a few boards, like qemu_x86, qemu_cortex_m3, native_posix, a couple of real boards - in parallel. Someone who doesn't test their code during development with multiple boards clearly doesn't do even the very minimal caring about portable code, and again, it's frightening to see the official Zephyr tool promoting such a crippled workflow.

Second issue is that there's no "west run". So, after an app is built, a user is thrown out in the code of CMake funkiness, like unknown/unexpected build system used, etc. The whole idea of west should be shielding a user from adhoc build systems like CMake, so all the common actions (build/run in emu/deploy/run in debugger) should be handled by it, or it's unclear why there's extra tool at all. 

All in all, I find current west is no replacement for a simple wrappers like zephyr-make: https://github.com/zephyrproject-rtos/zephyr/pull/5201 , which people with basic needs, like building for a number of boards in parallel, still have to use.

mbolivar, 2019-05-15:
> What it says there is that regardless of BOARD specified, it will always use one, the same build dir, or another words, at the same time, building for only one board is supported! 

That is by design. Most of our **users** are only going to build for one board. This keeps things simple. There is already an open issue discussing enhancements for Zephyr developers and maintainers: https://github.com/zephyrproject-rtos/west/issues/124; please feel free to comment there.

> It's actually frightening to think that west promotes such a workflow.

You are, of course, free to run west build multiple times with different arguments. Just like you can run cmake in multiple build directories. So I take it you have a similar issue open for Zephyr saying that cmake is not "adequate for real usage"?

> Someone who doesn't test their code during development with multiple boards clearly doesn't do even the very minimal caring about portable code, and again, it's frightening to see the official Zephyr tool promoting such a crippled workflow.

Wow. Just wow.

> Second issue is that there's no "west run". 

There is already an issue open for this: https://github.com/zephyrproject-rtos/zephyr/issues/5501

> All in all, I find current west is no replacement for a simple wrappers like zephyr-make: zephyrproject-rtos/zephyr#5201 , which people with basic needs, like building for a number of boards in parallel, still have to use.

Great! So since the concerns you've raised here are already covered by other issues, and you seem to have a workflow in place you are happier with anyway, I'm going to close this.

closed 2019-05-15 by mbolivar

mbolivar, 2019-05-15:
> Second issue is that there's no "west run".

BTW: `west build -t run`. This is in the Zephyr documentation.

pfalcon, 2019-05-15:
> That is by design

Seems like I was frightened for a reason ;-).

> So I take it you have a similar issue open for Zephyr saying that cmake is not "adequate for real usage"?

Absolutely, I consider CMake a regression to Zephyr build system usability (comparing to classic Zephyr versions), and opened the mentioned above https://github.com/zephyrproject-rtos/zephyr/pull/5201 in attempt to remedy that.

> and you seem to have a workflow in place you are happier with anyway

Noted that west developers suggest to not use west ;-).

> BTW: west build -t run

Well, that's not riddle enough ;-).

Thanks for response anyway!