zephyrproject-rtos/west #269: Escape tabs in west list -f

lemrey, 2019-05-24:
It would be nice if `west list -f` could escape tabs: e.g.
`west list -f "{revision} \t {name}"`

mbolivar, 2019-05-24:
I agree, this bugs me, too. On Unix you can do things like `west list -f "$(printf "{revision}\t\t{name}")"`, but that's clunky.

mbolivar, 2019-05-24:
Solutions:

- On bash, Control-V <TAB> enters a literal tab into Bash; this works fine in a west list format string if it's quoted.
- On bash, you can also write a tab with `$'\t'`, like this: `west list -f {name}$'\t'$'\t'{revision}`; I personally find that annoying, but maybe it's useful to others.
- In cmd.exe, there doesn't seem to be a good answer: https://serverfault.com/questions/210978/escape-tab-in-cmd-exe
- On Powershell, it seems like you can write \`t (backtick t) to escape a tab.

I wish cmd.exe didn't have this limitation, but since we can also pad with spaces (e.g. `west list -f {name:20} {revision}`), I think that's a workaround on Windows, so I'm going to close this.

mbolivar, 2019-05-24:
Also amusing is this megathread on how to create a tab in cmd.exe: https://stackoverflow.com/questions/10878138/how-to-create-tab-in-cmd

Amazing.

closed 2019-05-24 by mbolivar

lemrey, 2019-05-27:
Well, but these are really shell-dependent workarounds, wouldn't it be nice to have west support that?
I am having trouble getting a readable output from `west list`.

Padding does not seem to work:
```
$ west list -f "{revision:4}"
(not set)
5c5ba77d017071d7291d52e99d5460b1651e8de1
447495faa326082198c8e12493a2097f2ae20a99
ef1f9c3d87474ec3570b1f46e91fd4b54a4fb421
v0.4.0
c243b9a7a7b3c471023193992b46cf1bd1910450
031f3bbe45f8adf504ca3d13e6f093869920b091
mbedtls-2.13.1
```
The default output from `west list` does not look very readable:
```
$ west list
nrf            nrf                (not set)     (not set) (cloned)
fw-nrfconnect-zephyr zephyr             5c5ba77d017071d7291d52e99d5460b1651e8de1 https://github.com/NordicPlayground/fw-nrfconnect-zephyr (cloned)
fw-nrfconnect-mcuboot mcuboot            447495faa326082198c8e12493a2097f2ae20a99 https://github.com/NordicPlayground/fw-nrfconnect-mcuboot (cloned)
fw-nrfconnect-tinycbor modules/lib/tinycbor ef1f9c3d87474ec3570b1f46e91fd4b54a4fb421 https://github.com/NordicPlayground/fw-nrfconnect-tinycbor (cloned)
nrfxlib        nrfxlib            v0.4.0        https://github.com/NordicPlayground/nrfxlib (cloned)
cmock          test/cmock         c243b9a7a7b3c471023193992b46cf1bd1910450 https://github.com/ThrowTheSwitch/cmock (cloned)
unity          test/cmock/vendor/unity 031f3bbe45f8adf504ca3d13e6f093869920b091 https://github.com/ThrowTheSwitch/unity (cloned)
mbedtls        mbedtls            mbedtls-2.13.1 https://github.com/ARMmbed/mbedtls (cloned)
```

lemrey, 2019-05-27:
I could live with a workaround for my shell but that won't be enough unless padding works.

mbolivar, 2019-06-04:
> Padding does not seem to work:

```
$ west list -f '{revision:50} {name}'
(not set)                                          nrf
0023611bbfabd0698a38a12e6f30d7a2a9108702           fw-nrfconnect-zephyr
447495faa326082198c8e12493a2097f2ae20a99           fw-nrfconnect-mcuboot
ef1f9c3d87474ec3570b1f46e91fd4b54a4fb421           fw-nrfconnect-tinycbor
v0.4.0                                             nrfxlib
c243b9a7a7b3c471023193992b46cf1bd1910450           cmock
031f3bbe45f8adf504ca3d13e6f093869920b091           unity
mbedtls-2.13.1                                     mbedtls
$ west list -f '{revision:>50}'
                                         (not set)
          0023611bbfabd0698a38a12e6f30d7a2a9108702
          447495faa326082198c8e12493a2097f2ae20a99
          ef1f9c3d87474ec3570b1f46e91fd4b54a4fb421
                                            v0.4.0
          c243b9a7a7b3c471023193992b46cf1bd1910450
          031f3bbe45f8adf504ca3d13e6f093869920b091
                                    mbedtls-2.13.1
```

Working fine for me with west v0.5.8, and your example does what I would expect it to do. If you want truncation, use a period:

```
$ west list -f '{revision:.4}'
(not
0023
4474
ef1f
v0.4
c243
031f
mbed
```

lemrey, 2019-06-05:
Thank you Marti, truncation is what I needed and it works as expected! 🙇 s