zephyrproject-rtos/west #276: flash option to erase not working.

pabigot, 2019-06-18:
I'm sure I'm doing something wrong.

```
lilith[191]$ west --version
West bootstrapper version: v0.5.8 (/home/pab/.local/lib/python3.6/site-packages/west/_bootstrap)
West repository version: v0.5.8-8-gc23549f (/mnt/devel/external/zp/.west/west)
lilith[192]$ west flash -d xenon --flash-opt -e chip
ninja: no work to do.
Using runner: pyocd
usage: pyocd [-h] [--dt-flash {Y,y,N,n,yes,no,YES,NO}] --target TARGET
             [--daparg DAPARG] [--pyocd PYOCD] [--flash-opt FLASH_OPT]
             [--frequency FREQUENCY] [--gdb-port GDB_PORT] [--tui]
             [--board-id BOARD_ID]
pyocd: error: argument --flash-opt: expected one argument
lilith[193]$ west flash -d xenon --flash-opt -echip
ninja: no work to do.
Using runner: pyocd
usage: pyocd [-h] [--dt-flash {Y,y,N,n,yes,no,YES,NO}] --target TARGET
             [--daparg DAPARG] [--pyocd PYOCD] [--flash-opt FLASH_OPT]
             [--frequency FREQUENCY] [--gdb-port GDB_PORT] [--tui]
             [--board-id BOARD_ID]
pyocd: error: argument --flash-opt: expected one argument
lilith[194]$ west flash -d xenon --flash-opt '-e chip'
ninja: no work to do.
Using runner: pyocd
Flashing Target Device
usage: pyocd flash [-h] [-j PATH] [--config PATH] [--no-config]
                   [--script PATH] [-O OPTION=VALUE] [-da DAPARG [DAPARG ...]]
                   [--pack PATH] [-v] [-q] [-u UNIQUE_ID] [-b BOARD]
                   [-t TARGET] [-f FREQUENCY] [-W] [-e {auto,chip,sector}]
                   [-a ADDR] [--trust-crc] [--format {bin,hex,elf}]
                   [--skip BYTES]
                   PATH
pyocd flash: error: argument -e/--erase: invalid choice: ' chip' (choose from 'auto', 'chip', 'sector')
ERROR: command exited with status 2: pyocd flash -e sector -t nrf52840 '-e chip' /mnt/devel/zapps/mesh_sp/xenon/zephyr/zephyr.hex
run as "west -v flash -d xenon --flash-opt '-e chip'" for a stack trace
```

mbolivar, 2019-06-25:
```
--flash-opt -e chip
```

The shell is parsing `-e` as the value given to `--flash-opt`, and `chip` as a separate positional. Try quoting:

```
--flash-opt "-e chip"
```

pabigot, 2019-06-25:
That's the last example but using `'-e chip'` instead of `"-e chip"` , I think.

mbolivar, 2019-06-25:
Ah, sorry, I missed that there were multiple examples. I don't have a pyocd board on me right now but will get one later today to figure this out.

pabigot, 2019-06-25:
You don't need one; it fails before it gets to touching the hardware.

mbolivar, 2019-06-25:
```
west flash -d build/hello_world/particle_xenon/ --flash-opt "-e=chip"
```

This gets as far as failing with no debug probes for me -- but since I don't have a board, I'm not sure it's actually chip erasing.

mbolivar, 2019-06-25:
I'm sorry, wrong copy/paste.

```
west flash -d build/hello_world/particle_xenon/ --flash-opt=-e=chip
```

mbolivar, 2019-06-25:
An option to avoid the argparse-specific usage of `=` would be to split the `--flash-opt` argument before shelling out. Right now, it's passed in its entirety as a single argv option to the underlying pyocd call.

pabigot, 2019-06-25:
Thanks; that works.  I'll leave to you whether it's a bug that merits code changes; I wouldn't have thought of replacing spaces with `=`.

mbolivar, 2019-06-25:
> Thanks; that works. I'll leave to you whether it's a bug that merits code changes; I wouldn't have thought of replacing spaces with `=`.

At the very least we need to fix the help string. Let's start there and not do any splitting to keep backwards compatibility. I want to generalize the option and make it available to more runner backends, so we can deal with behavioral changes at that time.

closed 2019-06-25 by carlescufi