zephyrproject-rtos/west #278: RFC: support out-of-manifest manifest entries

pabigot, 2019-06-20:
May relate to #221.

What's below is a feature I think I want in order to simplify Zephyr development with unpublished versions of external projects, but I believe it could also contribute to a solution for foreign licensed material per recent TSC discussions.  (I'm in the camp that says "no non-Apache-2.0 material in default-installed Zephyr").

As I understand it West manifests are specified in a single file `west.yml`.  What I propose is to retain that, but also have a parallel directory, e.g. `west.d`, which contains additional files with `manifest` (and other?) sections.  Files in that directory (directly, not in subdirectories) should be processed in lexical order after processing `west.yml`, and if an entry matches the `name` of previously-encountered entry the new entry replaces the old one.

This approach would also allow Zephyr forks to be more resilient against changes to Zephyr's manifest contents: if I need a non-standard external repository I'd be describing it in a file that's independent of `west.yml`, making merges, rebases, and bisects significantly easier (no conflicts to resolve because upstream's file isn't modified).

For example, if upstream LVGL `lib/gui/lvgl` were to be registered in West rather than in `CMakeLists.lst` then I could more easily test Zephyr against in-progress enhancements to LVGL itself tracked in `west.d/lvgl.yml`, and publish that work in a branch that would be robust against upstream changes to `west.yml`.

To support material with a foreign license, processing files in `west.d` might recognize a `.ignore` suffix which signals they are not to be processed.  Opting in to the material would require the user to simply rename the file to remove the suffix.  If west generates some sort of flags that the build system can recognize then the glue code that is compatible with Apache-2.0 can be part of the repository but be disabled unless the `west.d/dirty.yml.ignore` (or `west.d/disabled/dirty.yml`) file was renamed.  People installing Zephyr from scratch will not contaminate their systems with material that is against their network and information systems policies, but if they want something they can easily enable it.

pabigot, 2019-06-20:
@carlescufi @microbuilder FYI

carlescufi, 2019-06-20:
>What I propose is to retain that, but also have a parallel directory, e.g. west.d, which contains additional files with manifest (and other?) sections.

I know that @mbolivar has a proposal he is working on that includes the `west.d` folder you mention in very similar terms, so we can consider this being addressed.

mbolivar, 2019-06-25:
> I know that @mbolivar has a proposal he is working on that includes the `west.d` folder you mention in very similar terms, so we can consider this being addressed.

https://github.com/mbolivar/zephyr/commit/a81aa4c81bc6dea3a0d25c47b5812f4dcbbadd20 is my latest hack at documenting the feature "as proposed". This is unimplemented.

Feedback is welcome!

pabigot, 2019-06-27:
> [mbolivar/zephyr@a81aa4c](https://github.com/mbolivar/zephyr/commit/a81aa4c81bc6dea3a0d25c47b5812f4dcbbadd20) is my latest hack at documenting the feature "as proposed". This is unimplemented.
> 
> Feedback is welcome!

That seems very powerful, once people wrap their heads around all the features.

It looks like it could do what I want as long as the Zephyr manifest includes:

    self:
        import: west-site.d

and an empty `west-site.d` directory, so I can do overrides without having to modify the Zephyr manifest.

mbolivar, 2019-06-28:
Thanks very much for your review. I agree that upstream zephyr ought to do something like that.

mbolivar, 2019-10-11:
Closing as a duplicate of #221

closed 2019-10-11 by mbolivar