zephyrproject-rtos/west #279: `west init -m /path/to/west.yml`

Blacksilver42, 2019-07-10:
Init from a local manifest file instead of always assuming it's a repository.
If the path specified on `-m` isn't a git repository, just copy it instead of cloning.

Allows for quickly testing a WIP manifest file without having to commit.

---

Alternatively, add a new flag that looks something like `-M MANIFEST_FILE`

tautologyclub, 2019-07-11:
To recap from Slack why I don't think it's a good idea: 

> I didn't say it was a really bad idea, I said it just might potentialyl encourage poor practices wrt version control while simultaneously offering extremely little in terms of developer convenience
> When contemplating adding a new feature to ANY software, the question really shouldn't be "why not"

Blacksilver42, 2019-07-11:
I'm creating a manifest repo from scratch; and if I want to test it, I have to commit.  
... and it probably won't work, so then I have to `git reset --hard HEAD^`, try again...

That's super tedious.

marc-hb, 2019-07-11:
Issues should describe the problem before suggesting solution(s). Based on the previous comment and some additional information on Slack, the real title of this issue seems to be something like "rapid manifest iteration is tedious because every change requires a commit", can you confirm?

@mbolivar is it correct that every manifest change requires a commit and if yes how come no one before @Blacksilver42 needed that?

tautologyclub, 2019-07-12:
I think a better solution would be to allow `--mr` to accept WORKING_TREE or something. This would at least mean west could tag it as such (think how `uname` gets the `-dirty` suffix when we have uncommitted changes when compiling the linux kernel). 

But really, `git commit --amend && west init ...`. Just make a one-liner out of it and the tedious part is reduced to nothing.

marc-hb, 2019-07-12:
> This would at least mean west could tag it as such (think how uname gets the -dirty suffix when we have uncommitted changes when compiling the linux kernel).

A bit off-topic: `git describe --dirty` does that. See example at https://github.com/zephyrproject-rtos/zephyr/pull/13608#pullrequestreview-206725920
(The implementation is actually a bit more complicated in the Linux kernel but exact same behavior https://lore.kernel.org/lkml/0dc31052-4cfd-3b92-0a4a-96c8ecfafff6@intel.com/ )

carlescufi, 2019-07-16:
@Blacksilver42 have you tried `west init -l myrepo/` ?

Blacksilver42, 2019-07-16:
@carlescufi Apparently not! It's still a bit tedious to make a copy of the "dirty" repo yourself, though.

marc-hb, 2019-07-16:
> It's still a bit tedious to make a copy of the "dirty" repo yourself, though.

Not clear to me why a copy of dirty repo helps, but if does then it's much more convenient to automate than ``git commit`. Real world example:
```
while true; do
  inotifywait -e close_write -r ~/masterrepo/
  time rsync -a --delete ~/masterrepo/ ~/repocopy/
  sleep 1
done
```
I use this to copy work in progress on remote systems and test multiple operating systems but `rsync` works just as well locally (BTW there are various reasons not to use `cp -r` ever, just use `rsync` always)

mbolivar, 2019-07-25:
> @mbolivar is it correct that every manifest change requires a commit 

Not if you've already initialized the west installation via west init.

I would do this as follows:

1. create a manifest repository with a stub manifest
2. `west init` using that
3. hack away inside the clone of the manifest repository created by `west init`. any `west` commands working on this clone do not require a commit. `west update` will use the working tree, e.g.
4. commit when ready

closed 2019-08-07 by Blacksilver42