zephyrproject-rtos/west #280: West build fails with link error 'multiple definition of main'

lawrencek52, 2019-07-15:
When a file is changed 'west build' fails to correctly link the project after rebuilding the changed file. The error is:

/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(main.c.obj): in function `main':
/home/lawrence/workspace/rc-demo/zephyr/zephyr/samples/basic/blinky/build/../src/main.c:18: multiple definition of `main'; app/libapp.a(main.c.obj):/home/lawrence/workspace/rc-demo/zephyr/zephyr/samples/basic/blinky/build/../src/main.c:18: first defined here
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/lawrence/workspace/rc-demo/zephyr/zephyr/samples/basic/blinky/build
run as "west -v build -b nrf52840_mdk" for a stack trace

To reproduce the issue:

cd zephyr/samples/basic/blinky
west build -b mdk52840_mdk
# west happily builds the west code and it works fine
touch src/main.c
west build -b mdk52840_mdk
# this time the build fails with a link error

marc-hb, 2019-07-16:
Thanks for the report. This can be reproduced with ninja without involving west at all, it's something I just broke in Zephyr's build system. Revert merged, should be working again now https://github.com/zephyrproject-rtos/zephyr/pull/17549
Once confirmed please close thanks!

lawrencek52, 2019-07-16:
> merged, should be working again now zephyrproject-rtos/zephyr#17549

I did 'west update' and the problem went away. Thanks

carlescufi, 2019-07-17:
Closing since solved

closed 2019-07-17 by carlescufi