zephyrproject-rtos/west #281: UnicodeDecodeError is raised while executing west flash

CliffsDover, 2019-07-18:
**Describe the bug**
`UnicodeDecodeError `is raised while executing `west flash`

```
  File "zephyrproject\.west\west\src\west\cmake.py", line 175, in load
    for line_no, line in enumerate(cache):
UnicodeDecodeError: 'cp950' codec can't decode byte 0xe2 in position 1838: illegal multibyte sequence
```

It comes from "zephyrproject\.west\west\src\west\cmake.py", line 175
I have to add `encoding="utf-8"` to line 174 to avoid this error

```
        with open(cache_file, 'r', encoding="utf-8") as cache:
            for line_no, line in enumerate(cache):
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr\samples\bluetooth\peripheral_hr
2. west build -b nrf52_pca10040
3. west flash

**Screenshots or console output**

```
\zephyrproject\zephyr\samples\bluetooth\peripheral_hr>west flash
-- west flash: rebuilding
ninja: no work to do.
Traceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python37\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 499, in main
    wrap(wrap_argv)
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 485, in wrap
    west.main.main(argv)
  File "zephyrproject\.west\west\src\west\main.py", line 580, in main
    args.handler(args, unknown)
  File "zephyrproject\.west\west\src\west\main.py", line 332, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "zephyrproject\.west\west\src\west\commands\command.py", line 90, in run
    self.do_run(args, unknown)
  File "zephyrproject\zephyr\scripts\west_commands\flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "zephyrproject\zephyr\scripts/west_commands\run_common.py", line 229, in do_run_common
    cache = cmake.CMakeCache(cache_file)
  File "zephyrproject\.west\west\src\west\cmake.py", line 170, in __init__
    self.load(cache_file)
  File "zephyrproject\.west\west\src\west\cmake.py", line 175, in load
    for line_no, line in enumerate(cache):
UnicodeDecodeError: 'cp950' codec can't decode byte 0xe2 in position 1838: illegal multibyte sequence
```

**Environment (please complete the following information):**
 - OS: Windows 10 Traditional Chinese
 - Toolchain: gnuarmemb
 - Commit SHA: 5113fec
 - Terminal: cmd.exe

mbolivar, 2019-07-25:
this should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/17769

carlescufi, 2019-07-25:
> this should be fixed by [zephyrproject-rtos/zephyr#17769](https://github.com/zephyrproject-rtos/zephyr/pull/17769)

Actually this was already fixed by https://github.com/zephyrproject-rtos/zephyr/pull/17637

closed 2019-07-25 by carlescufi