zephyrproject-rtos/west #283: Allow to specify repository name

vanwinkeljan, 2019-07-25:
It would be nice that there is an attribute to specify the repository name (eg. `repo-name`) that together with the `remote` can be combined into a URL to clone from.
This would allow to use the same repo for multiple projects, eg. in the case of multiple API version, without sacrificing the benefits of the remote concept.

See also: https://github.com/zephyrproject-rtos/zephyr/pull/17714#discussion_r307496625

mbolivar, 2019-07-25:
Thinking about it, that's reasonable and shouldn't be too hard to implement. I'm giving it a go now.

mbolivar, 2019-07-25:
@vanwinkeljan I've implemented this feature as `repo-path`, to make it clearer that it can contain slashes and is not just a name. Please see #284 -- any testing or feedback is welcome.

PR #284 (2019-07-26): West 0.6.0 rc2
Compared to rc1, this adds a bugfix and a new feature, the project 'repo-path' attribute, to allow continuing use of remotes with non-unique project names.

Fixes: #283
closed 2019-07-26 by carlescufi