zephyrproject-rtos/west #285: west init fails if ZEPHYR_BASE is set

mlennig, 2019-07-26:
[https://github.com/zephyrproject-rtos/west/issues/212](url)

Still an issue my dudes

mlennig, 2019-07-26:
```
Miriams-Air-622:ncs miriamlennig$ west init -m https://github.com/NordicPlayground/fw-nrfconnect-nrf
West already initialized. Aborting.
```

mbolivar, 2019-08-18:
Sorry for the long wait to reply.

TL;DR: this is intentional. If you have ZEPHYR_BASE set, it is used to find where the installation is located, which causes `west init` to believe you are already in one.

I don't like this at all but it was thrown in to support setups where people have `~/app` and `~/zephyrproject/.west` (etc.), i.e. applications entirely out of the west installation. In this case, running `west build` from `~/app` breaks unless there's some other way to tell west where zephyr is. We argued about how to handle this (I suggested adding a new `WEST_DIR` variable to tell west where to look), but another suggestion to use `ZEPHYR_BASE` won the day.

I was hesitant about this at the time and still think it's not great, since it ties west's multirepo core to Zephyr in an abstraction-breaking way. But history's inertia won as it usually does.

I'm trying to get to a future where, if you're using west, `zephyr-env.sh` is never needed, and setting ZEPHYR_BASE in the parent environment will only be something to do if you're not using west (or if you're doing something

Chandranshu9, 2020-12-22:
How can I resolve this issue?

mbolivar-nordic, 2021-01-05:
Unset ZEPHYR_BASE.

trond-snekvik, 2023-09-07:
Sorry to revive this old issue, but is there any way `west init` could just ignore this variable? There's no scenario where `ZEPHYR_BASE` is actually useful with `west init`, so `west init` just never works if you have set `ZEPHYR_BASE`. I can agree with the sentiment that `ZEPHYR_BASE` isn't the "right" way, to configure west, but it is documented as a valid workflow, so it's a bit unfortunate that it's incompatible with the most fundamental west command.

marc-hb, 2023-09-08:
> is there any way west init could just ignore this variable? 

This would make `west init` different from other `west` commands. Granted it is already different by design but it would make `west init` even more different, especially in the implementation.

> There's no scenario where ZEPHYR_BASE is actually useful with west init, so west init just never works if you have set ZEPHYR_BASE.

I don't think the second part is true. I wondered why #212 was marked as fixed and tested this. When there is no workspace where ZEPHYR_BASE points (for instance when someone is deleting and "re-initing"), then `west init` works4me. It does seem to ignore a non-existing workspace at ZEPHYR_BASE.

So I think `west init` fails only when ZEPHYR_BASE points to a valid workspace somewhere else. Can you please elaborate on that use cause, explain when it makes sense?

> but it is documented as a valid workflow, 

Barely?
- https://github.com/zephyrproject-rtos/zephyr/discussions/33521

marc-hb, 2023-09-11:
@mbolivar-ampere could you comment?