zephyrproject-rtos/west #288: west init fails

finikorg, 2019-08-06:
After cloning zephyr I run `west init` which fails.

```
=== Cloning west repository from https://github.com/zephyrproject-rtos/west, rev. master into zephyr/.west/west
Initialized empty Git repository in /tmp/2/zephyr/.west/west/.git/
remote: Enumerating objects: 111, done.
remote: Counting objects: 100% (111/111), done.
remote: Compressing objects: 100% (60/60), done.
remote: Total 2505 (delta 52), reused 71 (delta 33), pack-reused 2394
Receiving objects: 100% (2505/2505), 697.70 KiB | 1.65 MiB/s, done.
Resolving deltas: 100% (1442/1442), done.
From https://github.com/zephyrproject-rtos/west
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
 * [new tag]         v0.1.0     -> v0.1.0
 * [new tag]         v0.2.0     -> v0.2.0
 * [new tag]         v0.2.0rc1  -> v0.2.0rc1
 * [new tag]         v0.2.0rc2  -> v0.2.0rc2
 * [new tag]         v0.2.0rc3  -> v0.2.0rc3
 * [new tag]         v0.3.0     -> v0.3.0
 * [new tag]         v0.3.0rc1  -> v0.3.0rc1
 * [new tag]         v0.4.0     -> v0.4.0
 * [new tag]         v0.4.0rc1  -> v0.4.0rc1
 * [new tag]         v0.5.0     -> v0.5.0
 * [new tag]         v0.5.1     -> v0.5.1
 * [new tag]         v0.5.2     -> v0.5.2
 * [new tag]         v0.5.3     -> v0.5.3
 * [new tag]         v0.5.4     -> v0.5.4
 * [new tag]         v0.5.5     -> v0.5.5
 * [new tag]         v0.5.6     -> v0.5.6
 * [new tag]         v0.5.7     -> v0.5.7
 * [new tag]         v0.5.8     -> v0.5.8
 * [new tag]         v0.6.0     -> v0.6.0
 * [new tag]         v0.6.0rc1  -> v0.6.0rc1
 * [new tag]         v0.6.0rc2  -> v0.6.0rc2
 * [new tag]         v0.6.0rc3  -> v0.6.0rc3
b2ebf1e84824d88a9da7bffff9544ddeadd1c91d refs/remotes/origin/master
Branch 'master' set up to track remote branch 'master' from 'origin'.
Already on 'master'
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'post-init' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'config', 'selfupdate', 'help')
```

Blacksilver42, 2019-08-07:
Your west bootstrapper version is out of sync with the rest of west.

Dumb solution is to just install the entire latest version, which no longer has the whole bootstrapper thing. You'll have to update manually, though.
```
git clone https://github.com/zephyrproject-rtos/west
cd west
git checkout v0.6.0 # the tip is usually stable, but it might not be.
pip install .
```

Blacksilver42, 2019-08-07:
The `post-init` command was removed in cb660b6 along with the rest of the bootstrapper.

carlescufi, 2019-08-08:
Please update west with `pip3`.

closed 2019-08-08 by carlescufi

SIGSTACKFAULT, 2019-08-08:
oop, the version on pypi is 0.6.0. noted.