zephyrproject-rtos/west #289: How to use openocd with st-link for flashing nrf51?

Trubochkin, 2019-08-15:
Hello everyone!
I am trying to set up the environment for flashing NRF51 using STLINKv2. Unfortunately, I do not have an original J-Link programmer, but there is a STlink that works fine with OpenOCD.
I want to configure West together with OpenOCD so that I can use the `west flash --runner openocd` command to flash microcontroller. But I do not uderstand how to do it.
When I use the OpenOCD directly (without West ) I use the command:
`openocd -f "$env:OPENOCD_PATH\scripts\interface\stlink-v2.cfg" -f "$env:OPENOCD_PATH\scripts\target\nrf51.cfg" -c init -c "reset halt" -c "nrf51 mass_erase 0" -c "flash write_image nrf51_flash.hex" -c reset -c exit`

I would really appreciate your help!

mbolivar, 2019-08-17:
Hi @Trubochkin, have you taken a look at some of the other board.cmake files that use openocd?

Run `git grep openocd.board.cmake` in the zephyr repository to see how other board files do this.

Trubochkin, 2019-08-17:
Thanks for reply, @mbolivar. Yes, I looked at other files. But I still could not configure it. I'm definitely not sure, but it seems that the problem is either in [openocd.py](https://github.com/zephyrproject-rtos/zephyr/blob/08e0d1ff46dd93d9c4396336a40399e67f94cf0d/scripts/west_commands/runners/openocd.py) script or I did not pass the parameters to `board.cmake` file correctly. 
However, I no longer need this, because I have already set up environment commands in VScode IDE that execute OPENOCD and WEST independently of each other.

closed 2019-08-17 by Trubochkin