zephyrproject-rtos/west #297: Collect anonymous usage statistics

jhn-nordic, 2019-09-04:
Add anonymous usage statistics on which samples that are built and which libraries that are used.

This will make it possible to get more insight in how to use development resources on the things that matters most.

It is important that this is **_truly_** anonymous and that it **_only collect_** information about components that **_already are a part of the public repositories_**.

As west is open source it will be easy for users to inspect the implementation. 
It must also be easy and well documented how to turn this feature off.

E3V3A, 2019-09-04:
Come on, there is no such thing as "truly anonymous"! Anytime something is sent from some IP there is a record. At the very least, if you do decided to implement such a "feature", then it should be an optional non-default setting upon installation. (I.e. The user has to opt in for it to work.) 

Another issue, is that not everyone is connected all the time, which leaves even more room for weird failures, which are already abundant, AFAICT.

jhn-nordic, 2019-09-04:
I.e if this is enabled as a flag in west.yml then it could be default off and enabled by the manifest. The "server" it talks to could probably be defined there as well.

mbolivar, 2019-09-04:
@jhn-nordic @carlescufi I think this should be a noup patch in fw-nrfconnect-zephyr, not an upstream patch.

I'm also personally opposed to such an idea; I don't want to add a network dependency to `west build` and I'm one of those privacy weirdos that doesn't like my tools tracking me.

mbolivar, 2019-09-04:
> I.e if this is enabled as a flag in west.yml then it could be default off and enabled by the manifest. The "server" it talks to could probably be defined there as well.

In my opinion, this should be done using configuration options which default to "off, no tracking".

E.g. options `ncs.build.track-me`, `ncs.build.tracking-server`, or whatever.

I do not think this should go into the manifest syntax.

closed 2026-08-29 by marc-hb