zephyrproject-rtos/west #300: Make west posix compliant also for Cygwin

E3V3A, 2019-09-04:
As an avid Cygwin user, I like to to be able to also use Cygwin on all windows machines. 
Since Cygwin is almost completely POSIX compliant, I don't see a reason why west treating it as something weird. It's simply bash and python...

```bash
$ west --help

Traceback (most recent call last):
  File "/usr/bin/west", line 10, in <module>
    sys.exit(main())
  File "/cygdrive/c/cygwin64/lib/python3.8/site-packages/west/main.py", line 562, in main
    config.read_config()
  File "/cygdrive/c/cygwin64/lib/python3.8/site-packages/west/configuration.py", line 92, in read_config
    config.read(_gather_configs(configfile), encoding='utf-8')
  File "/cygdrive/c/cygwin64/lib/python3.8/site-packages/west/configuration.py", line 215, in _gather_configs
    ret.append(_location(ConfigFile.SYSTEM))
  File "/cygdrive/c/cygwin64/lib/python3.8/site-packages/west/configuration.py", line 191, in _location
    raise ValueError('unsupported platform ' + plat)
ValueError: unsupported platform CYGWIN_NT-6.3-9600
```

Can someone enlighten me as what problems this could create, if any?

mbolivar, 2019-09-04:
> Can someone enlighten me as what problems this could create, if any?
>
> Possibly related to #259.

Indeed it is! I added this code back when #259 was filed (a fix for which has been merged, but the user hasn't confirmed it, so I'm leaving it open for now) because I was trying to be conservative in the configuration code so as not to create or write to files outside of the west installation on platforms I'm not familiar with (e.g. with `west config --system section.key value`, which will create the system level config file if it does not exist).

Can you suggest a location for the system-level configuration file's path on Cygwin? I'd be happy to add support for it there, but I don't know how its file system mapping works and I understand that there's a bit of emulation and conversion going on.

E3V3A, 2019-09-04:
Hi @mbolivar ,

Yeah, I was looking and thinking. The problem is that people (like me) do weird things. 

So the question is, for `WEST_CONFIG_SYSTEM`, what is better?

* (a) That when user uses Cygwin it has it's own west config, in `$HOME/.westconfig`, **or**
* (b) when user uses Cygwin **or** Windows, it uses the same config (as Windows) in `%PROGRAMDATA%\\west\\config`

The code is something like: 
```python
...
    elif cfg == ConfigFile.SYSTEM:
        if 'WEST_CONFIG_SYSTEM' in os.environ:
            return os.environ['WEST_CONFIG_SYSTEM']

        plat = platform.system()
        if plat == 'Linux':
            return '/etc/westconfig'
        elif plat == 'Darwin':
            return '/usr/local/etc/westconfig'
        elif 'BSD' in plat:
            return '/etc/westconfig'
        elif 'CYGWIN_NT' in plat:
            return '~/.westconfig'                                      # Here we use the Cygwin user's $HOME
            #return os.path.expandvars('%PROGRAMDATA%\\west\\config')   # Here we use the Windows default %PROGRAMDATA%
        elif plat == 'Windows':
            return os.path.expandvars('%PROGRAMDATA%\\west\\config')
        else:
            raise ValueError('unsupported platform ' + plat)

...
```

This is still a little funny, because in many cases we don't have a system west config, and yet it will try to load from that path... anyway, that's where my cyg was crashing.

mbolivar, 2019-09-04:
So $HOME/.westconfig is the global config file, not the system one.

The system one is intended for use if there are multiple users on the same computer that want to share some configuration. That's why it's in /etc on Linux and BSD, e.g. It doesn't make sense to put the system config file in a user-specific directory.

I would want the default location for this file to be in the same place on Windows regardless of if the user was using Cygwin or cmd.exe or powershell or (... insert new shell here).

Does `os.path.expandvars("%PROGRAMDATA%\\west\\config)"` work in Cygwin too, and refer to the same file it would on cmd.exe? Or do you e.g. get posixpath for os.path in Python on Cygwin and have the `%PROGRAMDATA%` fail to be treated as an environment variable? And similarly for backslashed vs. forwards slashed paths.

mbolivar, 2019-09-04:
@E3V3A also, if you set WEST_CONFIG_SYSTEM to point to where you would like that file to be in your Cygwin environment, the above code will work.

E3V3A, 2019-09-05:
@mbolivar

> Does os.path.expandvars("%PROGRAMDATA%\\west\\config)" work in Cygwin too, and refer to the same file it would on cmd.exe?

Yeah, I don't see a clear way to get the raw Windows `%PROGRMADATA%`, but the mintty (bash) shell does add it as: `$ProgramData`. 

```bash
python3 -c "import os,platform;env=os.environ['ProgramData']; print(env);"
C:\ProgramData
```

E3V3A, 2019-09-05:
yes in fact setting it to anything: 
```bash
 WEST_CONFIG_SYSTEM=none west info
```
works.

mbolivar, 2019-09-05:
> Yeah, I don't see a clear way to get the raw Windows %PROGRMADATA%, but the mintty (bash) shell does add it as: $ProgramData. But it's not clear how general this is across different languages etc.

I like your idea of using `os.environ`. But I'm confused about the case difference (PROGRAMDATA vs. ProgramData). Are environment variables case insensitive in Cygwin? Is there a map from Cygwin variables to the underlying Windows ones?

>  WEST_CONFIG_SYSTEM=none west list

Be careful. If you write to any system config values, it will create a file named "none" in the current working directory and put them there. Not always what you want.

mbolivar, 2019-09-05:
BTW, besides `west config -h`, there's documentation here: https://docs.zephyrproject.org/latest/guides/west/config.html

E3V3A, 2019-09-07:
> Be careful. If you write to any system config values, it will create a file named "none" in the current working directory and put them there. Not always what you want.

Certainly not! We need a `/dev/null` for windows. 
Anyway, I'm getting along with `export WEST_CONFIG_SYSTEM=none` in my `.bashrc`. Let's see how that goes... So far there are no system configs...

Yes, I believe the mintty does "channel" native windows system variables, and changing their name a bit. I certainly have not added that. All Cygwins since a few years back are case sensitive. Unless you've installed cygwin on a non-ntfs insensitive drive...

mbolivar, 2019-09-10:
Sorry, I don't use cygwin so my questions are a bit naive here.

>> But I'm confused about the case difference (PROGRAMDATA vs. ProgramData). Are environment variables case insensitive in Cygwin? Is there a map from Cygwin variables to the underlying Windows ones?

> Yes, I believe the mintty does "channel" native windows system variables, and changing their name a bit. 

Are you saying that `os.environ['ProgramData']` in cygwin maps to the underlying Windows PROGRAMDATA environment variable, for both reads and writes?

If so, that's all I need to know to add a patch closing this issue.

If not, is there another key into os.environ I should be using instead?

Thanks!

E3V3A, 2019-09-11:
> Are you saying that os.environ['ProgramData'] in cygwin maps to the underlying Windows PROGRAMDATA environment variable? 

Yes.

> for both reads and writes?

No, there is no automatic or Cygwin native way to write Windows system environment variables, AFAIK. So if you need to **write**, you need to use a registry write, which can be done with:

1. Opening a CMD or PS sub-shell and use normal windows tools for registry operations
2. use sysinternals (?) `subinacl` when put in Cygwin PATH
3. use native Cygwin `regtool`.
 
```bash
$ subinacl.exe /help
SubInAcl version 5.2.3790.1180

USAGE
-----

Usage :
     SubInAcl [/option...] /object_type object_name [[/action[=parameter]...]

 /options    :
    /outputlog=FileName                 /errorlog=FileName
    /noverbose                          /verbose (default)
    /notestmode (default)               /testmode
    /alternatesamserver=SamServer       /offlinesam=FileName
    /stringreplaceonoutput=string1=string2
    /expandenvironmentsymbols (default) /noexpandenvironmentsymbols
    /statistic (default)                /nostatistic
    /dumpcachedsids=FileName            /separator=character
    /applyonly=[dacl,sacl,owner,group]
    /nocrossreparsepoint (default)      /crossreparsepoint

 /object_type :
    /service            /keyreg             /subkeyreg
    /file               /subdirectories[=directoriesonly|filesonly]
    /clustershare       /kernelobject       /metabase
    /printer            /onlyfile           /process
    /share              /samobject

 /action      :
    /display[=dacl|sacl|owner|primarygroup|sdsize|sddl] (default)
    /setowner=owner
    /replace=[DomainName\]OldAccount=[DomainName\]New_Account
    /accountmigration=[DomainName\]OldAccount=[DomainName\]New_Account
    /changedomain=OldDomainName=NewDomainName[=MappingFile[=Both]]
    /migratetodomain=SourceDomain=DestDomain=[MappingFile[=Both]]
    /findsid=[DomainName\]Account[=stop|continue]
    /suppresssid=[DomainName\]Account
    /confirm
    /ifchangecontinue
    /cleandeletedsidsfrom=DomainName[=dacl|sacl|owner|primarygroup|all]
    /testmode
    /accesscheck=[DomainName\]Username
    /setprimarygroup=[DomainName\]Group
    /grant=[DomainName\]Username[=Access]
    /deny=[DomainName\]Username[=Access]
    /sgrant=[DomainName\]Username[=Access]
    /sdeny=[DomainName\]Username[=Access]
    /sallowdeny==[DomainName\]Username[=Access]
    /revoke=[DomainName\]Username
    /perm
    /audit
    /compactsecuritydescriptor
    /pathexclude=pattern
    /objectexclude=pattern
    /sddl=sddl_string
    /objectcopysecurity=object_path
    /pathcopysecurity=path_container

Usage  : SubInAcl   [/option...] /playfile file_name

Usage  : SubInAcl   /help [keyword]
         SubInAcl   /help /full
    keyword can be :
    features  usage syntax sids  view_mode test_mode object_type
    domain_migration server_migration substitution_features editing_features
         - or -
    any [/option] [/action] [/object_type]

```

or with the Cgwin native:

```bash
$ regtool.exe --help
Usage: regtool [OPTION] ACTION KEY [data...]

View or edit the Win32 registry

Actions:

 add KEY\SUBKEY             add new SUBKEY
 check KEY                  exit 0 if KEY exists, 1 if not
 get KEY\VALUE              prints VALUE to stdout
 list KEY                   list SUBKEYs and VALUEs
 remove KEY                 remove KEY
 set KEY\VALUE [data ...]   set VALUE
 unset KEY\VALUE            removes VALUE from KEY
 load KEY\SUBKEY PATH       load hive from PATH into new SUBKEY
 unload KEY\SUBKEY          unload hive and remove SUBKEY
 save KEY\SUBKEY PATH       save SUBKEY into new file PATH
 restore KEY\SUBKEY PATH    restore SUBKEY from file PATH

Options for 'list' action:

 -k, --keys           print only KEYs
 -l, --list           print only VALUEs
 -p, --postfix        like ls -p, appends '\' postfix to KEY names

Options for 'get' action:

 -b, --binary         print data as printable hex bytes
 -n, --none           print data as stream of bytes as stored in registry
 -x, --hex            print numerical data as hex numbers

Options for 'set' action:

 -b, --binary         set type to REG_BINARY (hex args or '-')
 -d, --dword          set type to REG_DWORD
 -D, --dword-be       set type to REG_DWORD_BIG_ENDIAN
 -e, --expand-string  set type to REG_EXPAND_SZ
 -i, --integer        set type to REG_DWORD
 -m, --multi-string   set type to REG_MULTI_SZ
 -n, --none           set type to REG_NONE
 -Q, --qword          set type to REG_QWORD
 -s, --string         set type to REG_SZ

Options for 'set' and 'unset' Actions:

 -K<c>, --key-separator[=]<c>  set key-value separator to <c> instead of '\'

Options for 'restore' action:

 -f, --force    restore even if open handles exist at or beneath the location
                in the registry hierarchy to which KEY\SUBKEY points

Other Options:

 -h, --help     output usage information and exit
 -q, --quiet    no error output, just nonzero return if KEY/VALUE missing
 -v, --verbose  verbose output, including VALUE contents when applicable
 -w, --wow64    access 64 bit registry view (ignored on 32 bit Windows)
 -W, --wow32    access 32 bit registry view (ignored on 32 bit Windows)
 -V, --version  output version information and exit

KEY is in the format [host]\prefix\KEY\KEY\VALUE, where host is optional
remote host in either \\hostname or hostname: format and prefix is any of:
  root     HKCR  HKEY_CLASSES_ROOT (local only)
  config   HKCC  HKEY_CURRENT_CONFIG (local only)
  user     HKCU  HKEY_CURRENT_USER (local only)
  machine  HKLM  HKEY_LOCAL_MACHINE
  users    HKU   HKEY_USERS

If the keyname starts with a forward slash ('/'), the forward slash is used
as separator and the backslash can be used as escape character.
Example:
regtool list '/machine/SOFTWARE/Classes/MIME/Database/Content Type/audio\/wav'
```

E3V3A, 2019-09-11:
Or maybe using [python-registry](https://github.com/williballenthin/python-registry)...if compatible with Cygwins python interpreter. :laughing:

E3V3A, 2019-09-11:
Not easy to use, but something like: 
```bash
regtool list -v '/machine/SOFTWARE/Microsoft/Windows/CurrentVersion/Explorer/User Shell Folders/'

Common Startup (REG_EXPAND_SZ) = "%ProgramData%\Microsoft\Windows\Start Menu\Programs\Startup"
Common AppData (REG_EXPAND_SZ) = "%ProgramData%"
CommonPictures (REG_EXPAND_SZ) = "%PUBLIC%\Pictures"
Common Desktop (REG_EXPAND_SZ) = "%PUBLIC%\Desktop"
Common Start Menu (REG_EXPAND_SZ) = "%ProgramData%\Microsoft\Windows\Start Menu"
Common Programs (REG_EXPAND_SZ) = "%ProgramData%\Microsoft\Windows\Start Menu\Programs"
Common Templates (REG_EXPAND_SZ) = "%ProgramData%\Microsoft\Windows\Templates"

```

E3V3A, 2019-09-15:
Yeah, I hear you. This is quite messy if you wanna test this across other Windows platforms (using other languages, for example.) Seeing that I'm a fringe user, I'll just close this issue for now, and will get back if there are any new changes or revelations.

E3V3A, 2019-09-15:
Ok, I just noticed that your PR returns: 

```bash
# python3 -c "import os,platform;env=os.path.join(os.environ['ProgramData'], 'west', 'config'); print(env);"
C:\ProgramData/west/config
```

Now, this is probably ok in Cygwin as it does handle mixed Win/Posix paths...sometimes. But I'm against that, because if for some reason Cygwin tries to pass this to a native Win exec (like git or Go), it will probably error out.  Is there a way to tell python join to use a `\` as the join character?
(It does look good when using Windows python...)

mbolivar, 2019-09-16:
I'm guessing that cygwin trying to emulate POSIX as much as possible is causing python to use posixpath instead of ntpath as the os.path implementation, but the environment variable is using a backslash since it's coming up from Windows. Ugh.

>  if for some reason Cygwin tries to pass this to a native Win exec (like git or Go), it will probably error out

I've pushed another implementation which ought to return `C:\ProgramData\west\config`. Thanks for that.

mbolivar, 2019-09-16:
Reopening -- would you mind retesting? Thanks!

PR #309 (2019-10-05): configuration: add cygwin handling for the system config file
@E3V3A can you confirm this is a fix for you? It's as discussed in the issue you raised.

Fixes: #300
commit 635f0460 (2019-10-05): configuration: add cygwin handling for the system config file

GitHub user E3V3A reports that 'ProgramData' in the cygwin environment
corresponds to PROGRAMDATA in the parent environment, so we ought to
be able to use that to locate the system configuration file in a way
that is consistent with the outer native Windows shell expectations.

Fixes: #300
closed 2019-10-05 by carlescufi