zephyrproject-rtos/west #303: west init should show git commands/output at -v -v verbosity level

mbolivar, 2019-09-07:
All the other project related commands use west.manifest.Project.git under the hood, which logs git commands and their output at VERBOSE_VERY and above verbosity level.

West init just calls subprocess directly, since there is no manifest file to parse while it's running. However, it should still do the same thing as the other commands, to help users debug their init steps.

E3V3A, 2019-09-08:
I'm not sure what you're saying here, but I think west should show at least the executed git command when using `-v`. I think it would be very useful for debugging when on Windows platforms that may have multiple git installations. (e.g. user, system, chocolatey, cygwin etc.)

mbolivar, 2019-09-10:
For all commands *except* `west init`, west shows you the git commands at `-v -v` level. Note that `-v` is cumulative and can be given up to three times.

Try running `west -vv update` and you'll see what I mean. Then compare with `west -vv init` -- the git output isn't there. It should be for consistency, and that's what this issue tracks.

commit 55657a27 (2019-10-21): west init: print git commands at -v -v verbosity

Fixes: #303
closed 2019-10-21 by carlescufi