zephyrproject-rtos/west #314: manifest --freeze fails if a repository in the manifest is empty

skelliam, 2019-09-30:
My manifest has a git remote in it, but the remote repo is empty and not set up yet.
west update works fine.

west manifest --freeze will fail.

```
running 'git rev-parse refs/heads/manifest-rev' in C:\Users\skelliam\Projects\project\Application\App\Repo
fatal: ambiguous argument 'refs/heads/manifest-rev': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
'git rev-parse refs/heads/manifest-rev' in C:\Users\skelliam\Projects\project\Application\App\Repo finished with exit status 128 and wrote b'refs/heads/manifest-rev\n' to stdout
ERROR: command exited with status 128: rev-parse refs/heads/manifest-rev
Traceback (most recent call last):
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\main.py", line 583, in main
    args.handler(args, unknown)
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\main.py", line 300, in command_handler
    command.run(known_args, unknown_args, topdir, manifest=manifest)
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\commands\__init__.py", line 118, in run
    self.do_run(args, unknown)
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\commands\project.py", line 491, in do_run
    self._freeze(args)
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\commands\project.py", line 498, in _freeze
    frozen = Manifest.from_file().as_frozen_dict()
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\manifest.py", line 249, in as_frozen_dict
    sha = project.sha(QUAL_MANIFEST_REV_BRANCH)
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\manifest.py", line 668, in sha
    cp = self.git('rev-parse ' + rev, capture_stdout=True)
  File "C:\Users\skelliam\AppData\Local\Programs\Python\Python37\lib\site-packages\west-0.6.2-py3.7.egg\west\manifest.py", line 658, in git
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['rev-parse', 'refs/heads/manifest-rev']' returned non-zero exit status 128.
```

This is generally because rev-parse will fail on an empty repo.  You can try:
* mkdir foo
* cd foo
* git init
* git rev-parse HEAD  <-- fatal

mbolivar, 2019-10-01:
> My manifest has a git remote in it, but the remote repo is empty and not set up yet.
> west update works fine.

What do you mean by "empty"? West expects every project's remote to have at least one commit somewhere. I tried adding test coverage for this to west by adding a project without any commits in it to the manifest used in the test cases, but I got this error when I ran west update:

```
--- empty: fetching, need revision master
fatal: couldn't find remote ref master
fatal: the remote end hung up unexpectedly
```

Your steps to reproduce indicate that that's exactly what you tried, though, so I'm confused about how it could have possibly worked.

skelliam, 2019-10-03:
I should note that I'm not using west with Zephyr, I'm trying to use it for a different project.  The server is Bitbucket rather than git, but I suspect the underlying behavior is the same?

Here's what happens when I'm fetching the 'empty' repo as part of the manifest during west update:

```
=== updating EmptyRepo (Application\App\Location):
--- EmptyRepo : cloning and initializing
Initialized empty Git repository in C:/Users/skelliam/Projects/myproj/Application/App/Location/.git/
--- EmptyRepo : fetching, need revision master
fatal: couldn't find remote ref master
```

I also had 'fatal', yet west continued the update, and the repo was initialized as an empty repo.

skelliam, 2019-10-03:
At the end of the update, I do get an indication that the update failed for x, y, and z:

```
ERROR: update failed for projects: EmptyRepo, OtherEmptyRepo, YetAnotherEmptyRepo
```

However, at this point, west has already created the directory and initialized it with an empty repo.

skelliam, 2019-10-03:
Another note, my current manifest (this is a proof of concept for using west in our project) has 173 sub repos, so literally I minimize the cmd window while west update is happening, thus error messages like "couldn't find remove ref master" are lost on me.  The final ERROR message after everything is complete tells me there was something wrong, but it doesn't indicate that the project is in a state where it would crash west by attempting --freeze.

Multi-threaded support for update would be attractive but that's a side point.  :-)

mbolivar, 2019-10-03:
> I also had 'fatal', yet west continued the update, and the repo was initialized as an empty repo.
> [...]
> At the end of the update, I do get an indication that the update failed for x, y, and z:

Yep, this is expected. West keeps going and tries to do as much work as it can even if an individual project fails. You should have a non-zero exit code for the entire process to indicate the failure in addition to the indication of the error at the end of the update, so I wouldn't say that west update "worked" -- it does as much as it can, but it should be returning an error to your shell for the overall update operation. I'm not sure if windows captures this properly, but at least you get the error at the end.

> Another note, my current manifest (this is a proof of concept for using west in our project) has 173 sub repos, so literally I minimize the cmd window while west update is happening, thus error [...]

As you probably guessed, the final ERROR is intended to catch your intention exactly in this situation -- error occurs early, then scrolls off the screen. Something is "wrong".

In this case, the manifest really cannot be frozen: to freeze a manifest implies getting a SHA for each project. In this case, the EmptyRepo revision is "master", which is the default revision when one is not provided in the manifest file. Since this ref cannot be fetched from the remote (because the remote is empty), there is no SHA, and the manifest cannot sensibly be frozen. So the `west manifest --freeze` process failing is exactly what I would expect.

Now, that's not to say that you should be getting that nasty stack trace. I can add some robustness to `west manifest --freeze` to print a better error message if it can't get a SHA for the project revision, but:

- The update failed, and the --freeze cannot succeed for the reasons outlined above
- I can't think of a use case for an empty repository (as in no commits) being managed by west. If you need a repository with no files in the working tree, just use `git commit --allow-empty -m "initial empty commit"` in the remote repository and everything should work

skelliam, 2019-10-03:
Thanks, fair enough.  The root cause is I've split a large monorepo into many (170+) small repos.  I pushed all of the small repos to our Bitbucket server and then auto-generated a manifest.  The way that I tried to detect what was a repo or not a repo wasn't entirely robust. so I had some entries that were bogus.  So first workaround, is create at least one commit in the empty repos.
More robust handling would be nice so users don't experience a crash.

mbolivar, 2019-10-04:
>  More robust handling would be nice so users don't experience a crash.

Added in #318. Thanks for your report!

PR #318 (2019-10-04): manifest: improve Manifest.as_frozen_dict robustness
cc: @skelliam

In certain situations, such as if the project remote is an empty
repository or otherwise prevents 'west update' from setting up the
manifest-rev branch, we can't freeze the manifest even though a
project has been cloned.

Handle that by raising RuntimeError, so that higher layers know what
went wrong and can print a sensible error message instead of dumping
stack.

Fixes: #314
commit 5c109d67 (2019-10-04): manifest: improve Manifest.as_frozen_dict robustness

In certain situations, such as if the project remote is an empty
repository or otherwise prevents 'west update' from setting up the
manifest-rev branch, we can't freeze the manifest even though a
project has been cloned.

Handle that by raising RuntimeError, so that higher layers know what
went wrong and can print a sensible error message instead of dumping
stack.

Fixes: #314
closed 2019-10-04 by carlescufi