zephyrproject-rtos/west #322: `ninja flash` fails - 'ascii' codec can't decode byte 0xe2 in position 547

attie-argentum, 2019-10-09:
While getting my environment setup, I got the the point of running `ninja flash`, and saw the following output:

```
[0/1] Flashing stm32f723e_disco
Traceback (most recent call last):
  File "/usr/local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/west/main.py", line 583, in main
    args.handler(args, unknown)
  File "/usr/local/lib/python3.6/dist-packages/west/main.py", line 334, in ext_command_handler
    command.run(args, unknown, topdir, manifest=manifest)
  File "/usr/local/lib/python3.6/dist-packages/west/commands/__init__.py", line 118, in run
    self.do_run(args, unknown)
  File "/home/user/zephyrproject/zephyr/scripts/west_commands/flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "/home/user/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 229, in do_run_common
    cache = cmake.CMakeCache(cache_file)
  File "/usr/local/lib/python3.6/dist-packages/west/cmake.py", line 170, in __init__
    self.load(cache_file)
  File "/usr/local/lib/python3.6/dist-packages/west/cmake.py", line 175, in load
    for line_no, line in enumerate(cache):
  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 547: ordinal not in range(128)
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cd /home/user/zephyrproject/zephyr/samples/gui/lvgl/build && /opt/cmake/bin/cmake -E env /usr/local/bin/west flash --skip-rebuild
ninja: build stopped: subcommand failed.
```

I've resolved this with the following patch, applied in `/usr/local/lib/python3.6/dist-packages/west/`, though I suspect there could be more work required to resolve the issue "_properly_".

The `cache_file` in question is `/home/user/zephyrproject/zephyr/samples/gui/lvgl/build/CMakeCache.txt` (attached), which is indeed UTF-8... though I'm not sure it should be / needs to be...?

[image]

```patch
From c3ff49a61a2284124d7b407f12a799125584fa4c Mon Sep 17 00:00:00 2001
From: Attie Grande <attie.grande@argentum-systems.co.uk>
Date: Wed, 9 Oct 2019 15:03:19 +0000
Subject: [PATCH] fixed UTF-8 handling

---
 cmake.py | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/cmake.py b/cmake.py
index 37452e3..8fe5450 100644
--- a/cmake.py
+++ b/cmake.py
@@ -171,9 +171,9 @@ class CMakeCache:
 
     def load(self, cache_file):
         entries = []
-        with open(cache_file, 'r') as cache:
+        with open(cache_file, 'rb') as cache:
             for line_no, line in enumerate(cache):
-                entry = CMakeCacheEntry.from_line(line, line_no)
+                entry = CMakeCacheEntry.from_line(line.decode('utf-8'), line_no)
                 if entry:
                     entries.append(entry)
         self._entries = OrderedDict((e.name, e) for e in entries)
-- 
2.17.1
```

mbolivar, 2019-10-09:
Thank you for the report.

Zephyr master has moved this functionality out of west and into to
zephyr/scripts/west_commands/zcmake.py. The UTF-8 fixes
should be in there.

I see no reason why we shouldn't take a patch like this to this file for zephyr v1.14.x, though.

Eventually (a long time after that's not a supported LTS version), we'll
delete west.cmake entirely; it and west.build never should have been
there in the first place.

I think if you just propose a patch changing the open to use encoding
specifically you ought to get things working:

```
open(cache_file, 'r', encoding='UTF-8')
```

Mind sending a patch?

Thanks!

attie-argentum, 2019-10-09:
Hi, no problem - thanks for the quick response.

I didn't quite follow your comments on the new build system (I'm fairly new to Zephyr... and thought I was following the correct / current guide).

Happy to submit a patch though, and thanks for the tip re `encoding` parameter top `open()`.

commit c1faf365 (2019-10-09): resolved UTF-8 encoding issue in CMakeCache, closes #322
commit 2ea193df (2019-10-09): resolved UTF-8 encoding issue in CMakeCache, closes #322
closed 2019-10-09 by mbolivar