zephyrproject-rtos/west #332: west update will not fetch if tag is available locally

tejlmand, 2019-11-08:
During reviewing of https://github.com/NordicPlayground/fw-nrfconnect-nrf/pull/1437
I discovered that in case a tag is specified in manifest file and that tag is avaliable locally, then the remote will never be fetched.
https://github.com/zephyrproject-rtos/west/blob/f38b4dc1960eba5f14b359b0cef92dd417840e49/src/west/commands/project.py#L743

However, there is a small risk that a tag can be moved (bad practice I know, but it does happen)
or that two remotes has the same tag, but pointing to different SHAs (even smaller risk - when considering that tag must also be used in a manifest file, where SHAs are mostly used).

So this is a very minor issue, and not likely to happen, but if it happens that a tag is moved, then it might lead to confusion at end-users why their repo doesn't get updated with `west update`.

So mainly raising this so everyone is aware of this behavior and we can decide not to change it.
Feel free to comment.
```
west -vv update zephyr
ZEPHYR_BASE=/projects/github/ncs/zephyr (origin: env)
=== updating zephyr (zephyr):
running 'git rev-parse --show-cdup' in /projects/github/ncs/zephyr
'git rev-parse --show-cdup' in /projects/github/ncs/zephyr finished with exit status 0 and wrote b'\n' to stdout and wrote b'' to stderr
running 'git cat-file -t v2.0.99-ncs1-rc2' in /projects/github/ncs/zephyr
'git cat-file -t v2.0.99-ncs1-rc2' in /projects/github/ncs/zephyr finished with exit status 0 and wrote b'tag\n' to stdout and wrote b'' to stderr
skipping unnecessary fetch
```

mbolivar, 2019-11-12:
This is not a bug, it's a feature, and it's configurable :). You are merely describing the default behavior.

Please refer to the `update.fetch` config option (https://docs.zephyrproject.org/latest/guides/west/config.html#built-in-configuration-options) and `--fetch` option description in `west help update`.

mbolivar, 2019-11-12:
@tejlmand please consider closing this issue as it's not a bug -- or do you have some other suggestion for what to do with it? I think we don't need an open issue for this since this is the documented behavior.

tejlmand, 2019-11-18:
ahh, thanks.
Seems I haven't been updated on latest features ;-)
Closing.

closed 2019-11-18 by tejlmand