zephyrproject-rtos/west #333: Make west aware of when "west update" may be needed

jhedberg, 2019-11-08:
It'd be nice if west tracked the last time "west update" was run and compared that against the last modification timestamp of west.yml. That way west could either automatically run "west update" or issue a suggestion to the user that "west update" may be necessary.

mbolivar, 2019-11-08:
Interesting idea. Thanks for opening the issue!

Might be hard to get right with manifest imports, especially if importing a manifest from a project with a branch as its revision, instead of a SHA or a tag.

SebastianBoe, 2019-11-11:
This would help my workflow.

Usually, my workflow is like this:

```
git checkout new_zephyr_revision
build
# Get weird error
west update
build
# See error is fixed
```

Would be helpful if I could configure west to either auto-update or to error-out when the manifest does not match the git revisions (with a --force option to allow branches in zephyr modules). Or use the mechanism Johan suggests if that gets the job done better.

mbolivar, 2019-11-12:
I think this is a job for the build system, not west, actually.

Only the build system knows where ZEPHYR_MODULES ultimately came from and what its value was and why.

Like I said (perhaps too politely), I don't think it makes sense for west to do that if it's also going to allow manifest imports.

SebastianBoe, 2019-11-12:
Could, in the cases where ZEPHYR_MODULES is determined by west --list, west itself determine the modules and then tell CMake about them? Then we wouldn't be affected by the performance issue of loading west from CMake, and we would be able to satisfy this use-case, wouldn't we?

(for certain types of builds)

mbolivar, 2019-11-12:
I think that we would probably want to do that as a custom extension command, rather than using west list.

The contents of a module file is Zephyr-specific.

mbolivar, 2019-11-12:
Another thing to think about is the level of warning you might want. If this is for west build, you probably don't want to warn on a dirty tree.

However, when exactly to warn is going to be use-case specific:

- If I'm bisecting, I want to warn if HEAD != project.revision. While bisecting, HEAD is likely to be a descendant of project.revision if I've gone back in time in the zephyr repository. In this case, I want each repository to match zephyr/west.yml's revision for it exactly.
- If I'm just hacking on a module, I want to warn if `git merge-base --is-ancestor project.revision HEAD`. In this case, I probably explicitly want to allow modules to get ahead of west.yml, as otherwise I'll get warnings for local commits I want to be using.

mbolivar, 2019-12-19:
@jhedberg @SebastianBoe the more I think about this, the more convinced I am that this is a Zephyr build system issue.

I think what we need is for each Zephyr module to declare a semantic version (e.g. nrfx HAL version a.b.c), and for zephyr code that depends on it to declare a version range that it expects (e.g. nrf shims in zephyr/drivers say "I need at least x.y.z of nrfx").

This is the same thing we do with other out of tree dependencies such as host tools.

I can't think of a sane way to handle all the use cases discussed above within west proper.