zephyrproject-rtos/west #334: `west update` does not validate the origin of a SHA

tejlmand, 2019-11-13:
There is a potential flaw in how `west` handles SHAs that can potentially lead to unexpected behavior.

If a user updates the manifest file:
https://github.com/zephyrproject-rtos/zephyr/blob/f2cc6249158be67c77c8b1fd8ad4eb5d6d051b01/west.yml#L29, with a local SHA
```
projects:
    - name: hal_atmel
      revision: <Local only SHA>
      path: modules/hal/atmel
```
then `west update` will still work.
User at some point might commit and push this change remote, and for other users fetching this changed manifest, `west update` will fail, as the SHA is not remotely available.

Another side effect, could be when working with multiple upstream, as example working with both zephyr and NCS, I could add this to my manifest:
```
remotes:
      # nRF Connect SDK base URL.
    - name: ncs
      url-base: https://github.com/NordicPlayground
...
projects:
...
 - name: mcuboot
      repo-path: fw-nrfconnect-mcuboot
      revision: 135f716bfb9e2ffe926e3fed65bf825bf1d953da # SHA in Zephyr fork, NOT in NCS
      remote: ncs
...
```
and `west update` will work if I have that SHA already due to Zephyr manifest `west update`
https://github.com/zephyrproject-rtos/mcuboot/commit/135f716bfb9e2ffe926e3fed65bf825bf1d953da
but as that SHA is not available in NCS mcuboot fork, one might argue that `west update` should have failed with an error.

Now, having the possibility to change a SHA to something locally available is useful for local testing, but I think we should discuss if this is correct behavior.
Currently I see a couple of possibilities:
1) Document this behavior, and state this is intended
2) Ensure the SHA is present in the remote given, before checkout (will make testing will local SHA less convinient)
2b) Same as 2), but allow the keyword `remote: local` to keep the behavior in 1)

So one could write:
```
 - name: mcuboot
      repo-path: fw-nrfconnect-mcuboot
      revision: 135f716bfb9e2ffe926e3fed65bf825bf1d953da # SHA in Zephyr fork, NOT in NCS
      remote: local
```

Note: triggered by this: https://github.com/zephyrproject-rtos/west/pull/331#discussion_r345621585

tejlmand, 2019-11-13:
@carlescufi @mbolivar feel free to comment.
This is merely an open question, and I lean towards keeping the current behavior, even though I feel it is not fully correct behavior.

mbolivar, 2019-11-13:
> User at some point might commit and push this change remote, and for other users fetching this changed manifest, `west update` will fail, as the SHA is not remotely available.

I think that is a user error, personally. West doesn't push -- it only pulls. If we start to push, we have to increase the complexity of west in a way I don't see a clear benefit for in this case.

>     1. Document this behavior, and state this is intended
>     2\. Ensure the SHA is present in the remote given, before checkout (will make testing will local SHA less convinient)
>         2b) Same as 2), but allow the keyword `remote: local` to keep the behavior in 1)

I think it is important that west should work locally as much as possible. People get work done on airplanes, etc. Especially open source developers that go to a lot of conferences :).

I read through the west update documentation (https://docs.zephyrproject.org/latest/guides/west/repo-tool.html#main-commands) and the main thing I can see is that this step would need to be changed:

```
Fetches any project revisions in the manifest file which are not already pulled from the remote
```

Would you agree?

mbolivar, 2019-11-13:
> Would you agree?

In particular, I think we could change it to something like:

```
Fetches any project revisions in the manifest file which are not available locally. By default, tags and SHAs which are available locally are not fetched. Other revisions are assumed to be branches, which are considered to be stale by definition and will always be fetched.
````

tejlmand, 2019-11-18:
> In particular, I think we could change it to something like:
>
> Fetches any project revisions in the manifest file which are not available locally. By default, tags and SHAs which are available [...]

How about:
```
Checkout the revision specified in the manifest file.
If the revision cannot be found locally, then fetch the revision from the remote repository specified for the project.
Branches and full refs in the form of `refs/*`, `pull/*` are always fetched. 
```
would allow `west` to check locally before fetching.

mbolivar, 2019-11-18:
> Branches and full refs in the form of `refs/*`, `pull/*` are always fetched.

But that's not true, right? I think we want to document the current behavior, which is to fetch refs/heads/* into refs/west/*, but only if the revision might be a SHA.

marc-hb, 2019-11-21:
> User at some point might commit and push this change remote, and for other users fetching this changed manifest, west update will fail, as the SHA is not remotely available.

Isn't such a mistake instantly caught by even the most basic Continuous Integration?

It's basically a variant of forgetting to add a new file.

mbolivar-nordic, 2023-02-27:
IMO the current documentation is more than adequate (https://docs.zephyrproject.org/3.3.0/develop/west/built-in.html#west-update):

> Project update procedure:
>
> For each project that is updated, this command:
>
> 1. Initializes a local Git repository for the project in the workspace, if it does not already exist
> 2. Inspects the project’s revision field in the manifest, and fetches it from the remote if it is not already available locally
> 3. Sets the project’s manifest-rev branch to the commit specified by the revision in the previous step
> 4. Checks out manifest-rev in the local working copy as a detached HEAD
> 5. If the manifest file specifies a submodules key for the project, recursively updates the project’s submodules as described below.
>
> To avoid unnecessary fetches, west update will not fetch project revision values which are Git SHAs or tags that are already available locally. This is the behavior when the -f (--fetch) option has its default value, smart. To force this command to fetch from project remotes even if the revisions appear to be available locally, either use -f always or set the update.fetch configuration option to always. SHAs may be given as unique prefixes as long as they are acceptable to Git 1.
>
> If the project revision is a Git ref that is neither a tag nor a SHA (i.e. if the project is tracking a branch), west update always fetches, regardless of -f and update.fetch.
>
> Some branch names might look like short SHAs, like deadbeef. West treats these like SHAs. You can disambiguate by prefixing the revision value with refs/heads/, e.g. revision: refs/heads/deadbeef.

closed 2023-02-27 by mbolivar-nordic