zephyrproject-rtos/west #339: west: error: argument <command>: invalid choice: 'build' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'config', 'selfupdate', 'help')

emvalon, 2019-11-22:
I executed "west build -b qemu_x86 samples/hello_world" in my zephyr folder.
something wrong occured. 'build' is unsupported.
shown as below:
west: error: argument <command>: invalid choice: 'build' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'config', 'selfupdate', 'help')

closed 2019-11-23 by emvalon

NZSmartie, 2020-02-19:
I'm experiencing the same issue. 

1. To start with, I have an existing zephyr application that I'm updating to use the west ecosystem. 
   My app was provided with a `west.yml` containing the following example manifest:
   ```yaml
   manifest:
     remotes:
       - name: zephyrproject-rtos
         url-base: https://github.com/zephyrproject-rtos
     projects:
       - name: zephyr
         remote: zephyrproject-rtos
         revision: v2.2.0-rc1
         import: true
   ```
1. Then the manifest is initiated in the parent directory
    ```bash
    $ cd ..
    $ west init -l my-app
    # this takes time to checkout the repositories
    $ west update
    # Assuming all went well, I then attempt to build my application
    $ source zephyr/zephyr-env.sh
    $ cd my-app
    $ west build
    usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
    west: error: argument <command>: invalid choice: 'build' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'help', 'config', 'topdir', 'selfupdate')
    ````

It appears that west does not recognise the `build` command when a `west.yml` manifest is present. 

Is this intended behaviour, or is this a bug?

NZSmartie, 2020-02-19:
Sorry for reviving this issue. documentation is unclear about whether an application repository can also serve as the manifest repository

NZSmartie, 2020-02-19:
Turns out, there was a stray `.west/config` in my application repository. After deleting that, everything worked as expected. 

I hope if someone else has a similar issue, this may help them in the future