zephyrproject-rtos/west #340: What is the difference with tsrc?

arnaudgelas, 2019-11-25:
In the documentation [1], there are only `repo` and `git submodule` considered as reference.
Could you explain what is the difference or the benefit in using `west` rather than `tsrc` [2] ?

At first sight, it seems that there are lots of overlap in between `west` and `tsrc`.

Note that in [3], there are some comparisons with `repo`, `git submodule`, `git-subrepo`, `mu-repo`, `gr`

[1] https://docs.zephyrproject.org/latest/guides/west/why.html#rationale-for-a-custom-tool
[2] https://github.com/TankerHQ/tsrc
[3] https://tankerhq.github.io/tsrc/faq/

mbolivar, 2019-11-25:
To be honest, I had never heard of tsrc.

Do you have any experience with it? Would you be able to offer any comparison?

mbolivar-nordic, 2020-08-14:
No responses from the asker in a while, closing as stale. Thanks for your interest in west and please feel free to reopen.

closed 2020-08-14 by mbolivar-nordic