zephyrproject-rtos/west #341: west: west update brings different dependencies with same manifest

albertofloyd, 2019-11-28:
**Describe the bug**
Our project is using west manifest using start topology T2 as described in the documentation and following suggestions at https://github.com/zephyrproject-rtos/zephyr/issues/15144 and nordic example.

**To Reproduce**
1. cd ~
2. git clone https://github.com/albertofloyd/ec_west_test.git
3. cd ec_west_test
4. west init 
5. west list 
Here it shows zephyr-rtos projects
zephyr       zephyr                       N/A                                      N/A
hal_atmel    modules/hal/atmel            5690f5b84495c8b657ff204c5c827df1ab9e12e3 https://github.com/zephyrproject-rtos/hal_atmel
ci-tools     tools/ci-tools               c083680dd5e281218ddb47d0ceac5c5fd5619749 https://github.com/zephyrproject-rtos/ci-tools
https://github.com/zephyrproject-rtos/hal_openisa

If west init command is used to obtain the manifest instead, west list shows the right projects.
1.  cd ~
2. west init -m https://github.com/albertofloyd/ec_west_test.git
3. west list

ecfwwork     ecfwwork                     N/A                                      N/A
ec_west_test ecfwwork/manifest            master                                   https://github.com/albertofloyd/ec_west_test
basic-zephyr-ec ecfwwork/zephyr_ecfw         master                                   ssh://git@gitlab.devtools.intel.com:29418/wss-pse-io-and-power-delivery/basic-zephyr-ec
intel_boards ecfwwork/outoftreeboards     master                                   ssh://git@gitlab.devtools.intel.com:29418/wss-pse-io-and-power-delivery/intel_boards
zephyr_bsp_tests ecfwwork/bsp_tests           master                                   ssh://git@gitlab.devtools.intel.com:29418/wss-pse-io-and-power-delivery/zephyr_bsp_tests
zephyr       ecfwwork/zephyr              b2435d856a769f05310b8c6446efab105fd265bb https://github.com/zephyrproject-rtos/zephyr
hal_microchip ecfwwork/modules/hal/microchip 85302959c0c659311cf90ac51d133e5ce19c9288 https://github.com/zephyrproject-rtos/hal_microchip

**Expected behavior**
Same output from west list in both cases

mbolivar, 2019-12-02:
>     4\. west init

This is a user error. Plain west init with no arguments creates an installation in the current working directory using the default manifest, which is the upstream manifest, so the behavior you have descried is the expected behavior.

See `west init -h` for more details.

You want something like:

```
west init -m https://github.com/albertofloyd/ec_west_test.git my_workspace
cd my work_space
west update
```

closed 2019-12-02 by mbolivar

albertofloyd, 2019-12-02:
Suggested option works with remote manifest.

Actually what I wanted was to use a local manifest since this will be part of Zephyr-app.
Found this in the help as suggested 

git clone https://github.com/albertofloyd/app.git the_app
cd the_app
**west init -l** 
west update