zephyrproject-rtos/west #345: Branches with HEX names fails with `west update`

tejlmand, 2019-12-12:
If creating a branch where the name is a valid HEX number, and places this in the manifest file, then `west update` will fail.

As example, create a branch `deadbeef` and update the revision in the manifest to:
https://github.com/tejlmand/fw-nrfconnect-zephyr-1/tree/deadbeef

```
  projects:
    - name: zephyr
      url: https://github.com/tejlmand/fw-nrfconnect-zephyr-1
      revision: deadbeef
```

and the run `west update zephyr`

```
$ west update zephyr
=== updating zephyr (zephyr):
--- zephyr: fetching, need revision deadbeef
From https://github.com/tejlmand/fw-nrfconnect-zephyr-1
 * [new branch]            deadbeef                      -> refs/west/deadbeef
...
fatal: deadbeef: not a valid SHA1
ERROR: update failed for project zephyr
```

Workaround: specify full ref `refs/heads/deadbeef`:
```
--- zephyr: fetching, need revision refs/heads/deadbeef
From https://github.com/tejlmand/fw-nrfconnect-zephyr-1
 * [new branch]            deadbeef   -> refs/west/refs/heads/deadbeef
--- zephyr: checked out fb8467cde3f98a7be76ac0dd967480d2f5df2b99 as detached HEAD
```

https://github.com/zephyrproject-rtos/west/blob/master/src/west/commands/project.py#L727-L735

marc-hb, 2019-12-12:
`deadbeef~0` probably works too.

mbolivar, 2019-12-12:
> Workaround: specify full ref `refs/heads/deadbeef`:

Ah, good, you've already found the workaround I was going to suggest :).

> https://github.com/zephyrproject-rtos/west/blob/master/src/west/commands/project.py#L727-L735

Yep, that is why the function was renamed to _maybe_sha() from _is_sha() in 4d977988b172b270891a85293bac49e090eccf73. From the commit log:

```
Rename _is_sha() to _maybe_sha() -- it's technically possible to have
commit refs which are branches that look like SHAs. Any branch name
that begins with a letter A-F an[d] contains only hexadecimal characters
will work.
```

I don't think this is a bug, just an unavoidable ambiguity that may arise in ref parsing when we don't have the entire remote mirrored.

Do you have a suggestion for how to fix it, if you think it's a bug?

mbolivar, 2019-12-12:
> Yep, that is why the function was renamed to _maybe_sha() from _is_sha() in [4d97798](https://github.com/zephyrproject-rtos/west/commit/4d977988b172b270891a85293bac49e090eccf73)

Granted, that commit relaxed the requirement that the revision was 40 characters long to allow any <=40 character SHA to work, and it's probably a bit less likely to have an all-hex, exactly-40-character branch name than something like `deadbeef` :wink:.

But I think allowing unique SHA prefixes (like Git itself does) is a valuable feature, and the ambiguity has the workaround you've already found.

tejlmand, 2019-12-13:
> >    Yep, that is why the function was renamed to _maybe_sha() from _is_sha() in 4d97798
> 
> Granted, that commit relaxed the requirement that the revision was 40 characters long to allow any > <=40 character SHA to work

Seems you made use of the fact that I was away in August to let this slip in :stuck_out_tongue: 

> _maybe_sha() from _is_sha()

Yeah, I did noticed this, and was ... wasn't this named `_is_sha()` ?, and then just though.
I'm probably getting old, and it has always been named: `_maybe_sha()` :older_man: 

> Do you have a suggestion for how to fix it, if you think it's a bug?

Of course  :wink: .
Rough idea is to reintroduce the `len(s) == 40` for `_is_sha()` and then when we encounter `_maybe_sha()` then first try to fetch as a branch, and in case that fails, fetch as a SHA.

Currently, we we fetch everything when a SHA, we have actually fetched the branch and could try to check it out, and in case that fails, test to see if  the revision exists as a SHA.

But I would await the outcome of: https://github.com/zephyrproject-rtos/west/pull/344 as that could impact how much is fetched per default.

tejlmand, 2019-12-13:
> it's probably a bit less likely to have an all-hex, exactly-40-character branch name

Agree.

mbolivar, 2019-12-16:
> Rough idea is to reintroduce the `len(s) == 40` for `_is_sha()` and then when we encounter `_maybe_sha()` then first try to fetch as a branch, and in case that fails, fetch as a SHA.

I don't get it, sorry.

Are you saying we should treat everything except 40-character SHAs as if they were branches?

That would mean people could no longer use abbreviated SHAs as revisions.

tejlmand, 2019-12-18:
> I don't get it, sorry.
>
> Are you saying we should treat everything except 40-character SHAs as if they were branches?
>
> That would mean people could no longer use abbreviated SHAs as revisions.

I'm saying that we could safely assume that a 40-character hex string is a SHA.

And that in case we encounter a <40 character hex string we cannot know if that is a SHA or a branch/tag. (assuming we don't have it locally already).

A branch/tag can be fetched directly, so we can try to do:
```
git fetch <url> deadbeef
fatal: Couldn't find remote ref deadbeef
```
this failed, then it's most likely a SHA, and we fetch whole ref-space, just as today when a SHA is provided.

Note: we should of course test to see if the hex-string is available in local ref space before doing all of this. (smart fetching)

mbolivar, 2019-12-19:
> this failed, then it's most likely a SHA, and we fetch whole ref-space, just as today when a SHA is provided.

OK, that makes sense.

We can skip this fallback if the user provides `revision-in` as well.

mbolivar-nordic, 2021-04-20:
I think this should be a `wontfix` because such branches can always be disambiguated with `refs/heads/deadbeef` style naming, and being able to type short names is really worthwhile for UX.

closed 2021-04-20 by mbolivar-nordic