zephyrproject-rtos/west #347: Consider using pygit2 instead of shelling out to git

mbolivar, 2019-12-20:
I wonder if we should make west rely on pygit2 instead of shelling out to git internally.

- pro: new versions of git (like 2.24) shouldn't break us, because pygit2 uses libgit2 behind the scenes
- pro: potential speed improvements on windows, since we might avoid subprocess creation (slow `west update` on windows is one of the main reasons I can't make myself use that OS day to day)
- con: big rewrite

pygit2 just hit version 1.x and has a backwards compatibility policy. libgit2 is what's used by major git forges to do git operations, so it's battle tested.

mbolivar, 2019-12-20:
@carlescufi @tejlmand @marc-hb comments welcome

mbolivar, 2019-12-20:
I'm not actually suggesting we do this; I just want to have a place to discuss the idea. I'm currently against it as too much effort for not enough clear benefit.

marc-hb, 2019-12-20:
> con: potential big loss of debuggability (e.g. can't strace to see what git operations are happening)

I don't know anything about pygit2; if it's successful I would expect it to have decent logging? If not then yeah probably better forget about it.

> con: big rewrite

There's already (for obvious reasons) a git helper/indirection in `manifest.py#Project.git()`, wouldn't that make the rewrite mostly contained there hence easier?

mbolivar, 2019-12-20:
>  don't know anything about pygit2; if it's successful I would expect it to have decent logging? If not then yeah probably better forget about it.

It's just an FFI wrapper around https://libgit2.org/.

> Tere's already (for obvious reasons) a git helper/indirection in `manifest.py#Project.git()`, wouldn't that make the rewrite mostly contained there hence easier?

Mostly the other specific methods would be easier to wrap. That lowest-level one is just a way to call git itself.

mbolivar, 2020-02-25:
Closing; answer is "no"

closed 2020-02-25 by mbolivar