zephyrproject-rtos/west #349: 3ac600ac breaks fetches from tags

mbolivar, 2020-01-13:
Commit 3ac600acaa112fb5d7edfcfd9f1873ed2edc516c breaks setting a revision to a tag.

For example, with https://github.com/mbolivar/my-zephyr-app/blob/202a5ffba528e95255e8c62de919c7510db349ad/west.yml, I get:

```
$ west update
=== updating zephyr (zephyr):
--- zephyr: cloning and initializing
Initialized empty Git repository in /tmp/zephyr-imports/zephyr/.git/
--- zephyr: fetching, need revision v2.1.0
remote: Enumerating objects: 394472, done.
remote: Total 394472 (delta 0), reused 0 (delta 0), pack-reused 394472
Receiving objects: 100% (394472/394472), 284.55 MiB | 8.41 MiB/s, done.
Resolving deltas: 100% (294858/294858), done.
error: cannot update ref 'refs/heads/manifest-rev': trying to write non-commit object a8a050b7385f5c9c5478412f143f45d8e90045f2 to branch 'refs/heads/manifest-rev'
From https://github.com/zephyrproject-rtos/zephyr
 ! [new tag]         v2.1.0         -> manifest-rev  (unable to update local ref)
[...snip...]
https://github.com/mbolivar/my-zephyr-app/blob/202a5ffba528e95255e8c62de919c7510db349ad/west.yml
```

mbolivar, 2020-01-13:
@tejlmand I am working on a fix. No action on your side is requested.

tejlmand, 2020-01-14:
> 
> 
> @tejlmand I am working on a fix. No action on your side is requested.

great.
Probably that's why the fetching was done first, and then do `update-ref`

Just restore old behavior for that part.

tejlmand, 2020-01-14:
https://github.com/zephyrproject-rtos/west/pull/331#discussion_r355721212
> Fetching directly into manifest-rev leaves a nice clean entry in my reflog:

So now we know why this wasn't done that way originally.

closed 2020-01-14 by mbolivar

mbolivar, 2020-01-14:
> So now we know why this wasn't done that way originally.

Indeed. Sorry for all the bother.

PR #350 (2020-01-14): Tag fixes
The main point of this PR is commit `update: fix _fetch() when revisions are tags`, which:

Fixes: #349

Everything else is just scaffolding and regression testing.
commit 2287f174 (2020-01-14): update: fix _fetch() when revisions are tags

Whoops. I should have thought of this when requesting this change.

When a revision is not a SHA, fetch it and update manifest-rev to it
in two steps, setting manifest-rev to FETCH_HEAD^{commit}.

This still only fetches the refs we need, but works for all types of
refs.

Fixes: #349