zephyrproject-rtos/west #365: Support git submodules

mbolivar, 2020-01-31:
This issue tracks adding support for projects with git submodules to west.

The overall goal is to clone and update submodules on `west update` in addition to the projects that contain them.

finikorg, 2020-07-03:
We have now project we want to add as a Zephyr module. It has submodule. At the moment we are investigating what is the best way of handling it. Support git submodules in west would help.

carlescufi, 2020-11-27:
I suggest something like this in the manifest schema:

```
  projects:
    required: true
    type: seq
    sequence:
      - type: map
        mapping:
          # Project's unique name.
          name:
            required: true
            type: str
          submodules:
             required: false
             type: any
```

and then do something similar to what we have in the `import:` field so you could do:

```
submodules: true
```
or
```
submodules:
    name-allowlist:
        - submodule_1
        - submodule_2
```
or
```
submodules:
    name-denylist:
        - submodule_1
        - submodule_2
```

mbolivar-nordic, 2020-12-01:
I think I would prefer to start with a `submodules: true` (which defaults to `false`) to update the submodules as well with `git submodule update` or so, before we go crazy with allow or blocklists.

marc-hb, 2021-01-08:
@finikorg would you have time to give #459 a try?

tejlmand, 2021-01-11:
> I think I would prefer to start with a `submodules: true` (which defaults to `false`) to update the submodules as well with `git submodule update` or so, before we go crazy with allow or blocklists.

There is already a use-case where `west` is used to clone a single repo, which also happens to be a one submodule out of two.
https://github.com/nrfconnect/sdk-nrf/blob/master/west.yml#L107-L114
```
    - name: cmock
      path: test/cmock
      revision: c243b9a7a7b3c471023193992b46cf1bd1910450
      remote: throwtheswitch
    - name: unity
      path: test/cmock/vendor/unity
      revision: 031f3bbe45f8adf504ca3d13e6f093869920b091
      remote: throwtheswitch
```
where the `cmock` repo actually contains two submodules:
- `c_exception`
- `unity`

so I think there is a use-case for specific list of submodules to clone, in addition to all. (when `true` is given)

closed 2021-01-11 by mbolivar-nordic