zephyrproject-rtos/west #367: Add 'directory' key in west import map

mbolivar, 2020-02-03:
Example syntax:

```
manifest:
  projects:
  - name: zephyr
    url: https://github.com/zephyrproject-rtos/zephyr
    import:
      directory: zephyr_projects
```

Semantics:

- the `zephyr` project, and all imported projects, have `zephyr_projects` as an additional path component. e.g. `zephyr_projects/zephyr` instead of `zephyr` as the path for `zephyr` project
- the `directory` value cannot escape the workspace, e.g. `directory: ../..` is an error
- the `directory` value is cumulative, e.g. if the imported manifest in turn does an import with `directory: foo`, then the combined `directory` is `zephyr_projects/foo`

Alternate suggested spellings for `directory`:

- `path`
- `path-prefix`

mbolivar, 2020-02-03:
cc @henrikbrixandersen @tejlmand

henrikbrixandersen, 2020-02-03:
LGTM. Personally I prefer `directory`.

DenvE, 2020-02-10:
Directory is fine fo me too

marc-cpdesign, 2020-02-19:
I'd love to see this implemented also !

I'll take whatever keyword I get for 'directory', but also consider some alternatives:
- folder
- destination
- destination-prefix
- base-prefix
- base-destination

(Having 'prefix' or 'base' in there appeals to me because the _actual_ destination is the path+name)

marc-hb, 2020-02-19:
> (Having 'prefix' or 'base' in there appeals to me because the actual destination is the path+name)

Agreed, "directory" or "path" are super generic terms, way too vague when left alone.

Inspired by "destination" and trying to describe the action rather than the elusive nature, how about `into:`?

```
manifest:
  projects:
  - name: zephyr
    url: https://github.com/zephyrproject-rtos/zephyr
    import:
      into: all_imports/zephyr_clones/
```
By the way will this new prefix support more than one multiple directory level?

mbolivar, 2020-02-19:
> By the way will this new prefix support more than one multiple directory level?

It should IMO.

marc-cpdesign, 2020-02-19:
+1 for "into"
+1 for recursive/ multi-level import support

Let me know if I can help somehow.

euripedesrocha, 2020-04-24:
I would like to have something like this. But I would prefer a syntax that adds a prefix to the defaults section of the manifest:
```yaml
defaults:
    path_prefix: prefix_path
```
My goal at this moment is to have only the projects I´m using by setting import as false.

tejlmand, 2020-04-30:
Just a thought.
Today I can have the following in a manifest:

```
manifest:
  projects:
  - name: zephyr-project
    path: zephyr-upstream
    repo-path: zephyr
    remote: upstream
```
and now you suddenly add `directory` as a keyword, so a manifest could end up as:
```
manifest:
  projects:
  - name: zephyr-project
    path: zephyr-upstream
    repo-path: zephyr
    remote: upstream
    import:
      directory: zephyr_projects
```

mixing `path` and `directory` in our yaml is inconsistent, so simply for that reason I dislike directory.

In this case, `import-path` or `path-prefix` would fit better in current structure, as well as little more self-explaining.

And we would end up with:
```
manifest:
  projects:
  - name: zephyr-project
    path: zephyr-upstream
    repo-path: zephyr
    remote: upstream
    import:
      import-path: zephyr_projects
      path-prefix: zephyr_projects
```

marc-hb, 2020-05-04:
> mixing path and directory in our yaml is inconsistent,

While I find path and directory both too generic, I think the difference between them is well defined: a path is (long story short) a list of nested directories. This definition is surprisingly consistent across languages:
https://docs.python.org/3/library/pathlib.html#accessing-individual-parts
https://perldoc.perl.org/File/Path.html
https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.management/split-path
https://docs.oracle.com/javase/8/docs/api/java/nio/file/Path.html
https://en.cppreference.com/w/cpp/filesystem/path

Hence my earlier question about supporting multiple levels.

tejlmand, 2020-05-04:
> > mixing path and directory in our yaml is inconsistent,
> 
> While I find path and directory both too generic, I think the difference between them is well defined: a path is (long story short) a list of nested directories. This definition is surprisingly consistent across languages:

I was more referring to the situation in the original proposal where the yaml could look as:
```
manifest:
  projects:
  - name: zephyr-project
    path: zephyr-upstream
    import:
      directory: zephyr_projects
```
in this case I personally fail to see why `path` is correct under `projects`, and `directory` would be correct under the `import`.
As I assume users can write:
```
    import:
      directory: imported_projects/zephyr_projects
```
then it seems to my the `path`-based keyword is still most correct, and what is already in use.

mbolivar-nordic, 2020-08-14:
@marc-hb  and @tejlmand have convinced me that the keyword should contain 'path'. I think I'll use 'path-prefix'.

I added #423 for this. Testing and feedback very much desired.

closed 2020-08-14 by mbolivar-nordic