zephyrproject-rtos/west #373: Error: unexpected keyword argument 'requires_workspace'

erwango, 2020-02-07:
I'm seeing this error on automated test bench.
```
$ west update
Traceback (most recent call last):
  File "/local/home/frq07517/.local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/west/app/main.py", line 767, in main
    app = WestApp()
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/west/app/main.py", line 64, in __init__
    lst = [cls() for cls in classes]
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/west/app/main.py", line 64, in <listcomp>
    lst = [cls() for cls in classes]
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/west/app/project.py", line 153, in __init__
    requires_workspace=False)
TypeError: __init__() got an unexpected keyword argument 'requires_workspace'
```

FWIW `python3.6 -m pip install --user -r $ZEPHYR_BASE/scripts/requirements.txt` is done at each run.
I have to manually revert to 0.6.3 verion with `pip3 install --user -U west==0.6.3` to get it working back.

erwango, 2020-02-07:
Tried to uninstall and install back and errors shows up again:
```
$ pip3 uninstall west
Uninstalling west-0.7.0:
  /local/home/frq07517/.local/bin/west
  /local/home/frq07517/.local/lib/python3.6/site-packages/west-0.7.0.dist-info/INSTALLER
  /local/home/frq07517/.local/lib/python3.6/site-packages/west-0.7.0.dist-info/LICENSE
```
And install back, but no luck ...

erwango, 2020-02-07:
`rm -rf /local/home/moi/.local/lib/python3.6/site-packages/west` did the trick. Thanks @nashif .

@mbolivar we can close the point unless something could be done on west side

mbolivar, 2020-02-07:
I couldn't reproduce this on Arch Linux, but I can on Ubuntu 18.04.

On Ubuntu, if I do:

```
pip3 install west==0.6.3 # note --user is added by default on ubuntu
pip3 install west==0.7.0
```

Then I see the following:

```
$ ls ~/.local/lib/python3.6/site-packages/west
__init__.py  app           build.py  commands     configuration.py  main.py              manifest.py  version.py
__pycache__  backports.py  cmake.py  commands.py  log.py            manifest-schema.yml  util.py      west-commands-schema.yml
```

This is wrong: `commands.py` and `commands` the directory should not both be there. West 0.7.0 has a `commands.py`, while west 0.6.x have a `commands` directory.

So something is wrong with the upgrade procedure for this version of pip. OK, `apt-cache show python3-pip` says `Version: 9.0.1-2`, but if I try installing pip version `9.0.1` on Arch in a virtual environment then repeating the above steps to install 0.6.3 then upgrade to 0.7.0, I see:

```
app       cmake.py     configuration.py  log.py       manifest-schema.yml  util.py     west-commands-schema.yml
build.py  commands.py  __init__.py       manifest.py  __pycache__          version.py
```

No `commands` directory, which is what I expect.

So at this point I'm wondering if something is wrong with Ubuntu 18.04's extra patches (implied by the `-2` suffix) on top of pip 9.0.1.

mbolivar, 2020-02-07:
Closing since the workarounds are reported to fix the problem

closed 2020-02-07 by mbolivar