zephyrproject-rtos/west #378: west init -l does not set zephyr.base

tejlmand, 2020-02-11:
`west init -l` in west 0.7.1 does not set `zephyr.base` in `.west/config` which causes following west commands, such as build and flash to fail.

```
$ west build -bnrf52840_pca10056 
WARNING: can't find the zephyr repository
  - no --zephyr-base given
  - ZEPHYR_BASE is unset
  - west config contains no zephyr.base setting
  - no manifest project has name or path "zephyr"

  If this isn't a Zephyr workspace, you can   silence this warning with something like this:
    west config zephyr.base not-using-zephyr
-- west build: build configuration:
       source directory: /tmp/pr-test-ncs/22707/zephyr/zephyr/samples/hello_world
       build directory: /tmp/pr-test-ncs/22707/zephyr/zephyr/samples/hello_world/build (created)
       BOARD: nrf52840_pca10056 (origin: command line)
-- west build: generating a build system
CMake Error at CMakeLists.txt:5 (include):
  include could not find load file:

    /cmake/app/boilerplate.cmake
```

tejlmand, 2020-02-11:
Error introduced in this commit:
0831facadf35a52d9e4a23caf808a0c62e5aa50e

tejlmand, 2020-02-11:
also happens if you just do: `west init`

Content of `.west/config` after init:
```
[manifest]
path = zephyr
```

mbolivar, 2020-02-11:
What do you think about just setting this to medium? There is an easy workaround; 'west config zephyr.base zephyr'.

tejlmand, 2020-02-11:
As it may happen to any user who just do:
```
mkdir /<path-to-new-project>/
west init
cd zephyr/samples/hello_world
west build -b<some_board>
```
then I think it's high prio, cause it's not a good first time user experience.

Also build servers may be affected, and I don't think people like to start adding this config everywhere.

PR #379 (2020-02-13): zephyr base: Fixes issue with failing to discover Zephyr base
Fixes: #378

When Zephyr is also the manifest repository then west fails to identify
ZEPHYR_BASE correctly as the project name is manifest, with path zephyr.

Therefore, before checking projects with path `zephyr` we first check if
the manifest repository has path == zephyr.
commit e0224348 (2020-02-13): zephyr base: Fixes issue with failing to discover Zephyr base

Fixes: #378

When Zephyr is also the manifest repository then west fails to identify
ZEPHYR_BASE correctly as the project name is manifest, with path zephyr.

Therefore, before checking projects with path `zephyr` we first check if
the manifest repository has path == zephyr.
commit 471328db (2020-02-17): zephyr base: Fixes issue with failing to discover Zephyr base

Fixes: #378

When Zephyr is also the manifest repository then west fails to identify
ZEPHYR_BASE correctly as the project name is manifest, with path zephyr.

Therefore, before checking projects with path `zephyr` we first check if
the manifest repository has path == zephyr.
closed 2020-02-13 by mbolivar