zephyrproject-rtos/west #383: West commands not working in terminal with version v.0.7.1

BrennaBelletti, 2020-02-14:
I installed the latest version of west (v.0.7.1) with pip3 like follows:

`pip3 install west`

I then run west --version and get the following error: 

`-bash: west: command not found`

This is true for any west commands I try to run.

I spent a while trying to debug the issue, but then decided to revert to a previous version. 

I ran 

```
pip3 uninstall west
pip3 install west==0.7.0
```

Now running `west --version` I get the proper output, and all west commands work. 

`West version: v0.7.0`

I am using iTerm2 on macOS Mojave.

marc-h38, 2020-02-14:
Changing your PATH is required to use any `pip3` program, this is documented at https://docs.zephyrproject.org/latest/getting_started/index.html

mbolivar, 2020-02-14:
> Changing your PATH is required to use any `pip3` program, this is documented at https://docs.zephyrproject.org/latest/getting_started/index.html (admittedly not the best place if you don't use Zephyr)

This is actually operating system dependent. Several Linux distributions more recent than Ubuntu 18.04 put ~/.local/bin in PATH automatically when using pip3, for example.

@BrennaBelletti -- can you please make sure the directory that the west binary is installed into is in your PATH?

You can find it in the `pip3 show -f west` output.

BrennaBelletti, 2020-02-14:
Yea, before the executable for the latest version wasn't being put into my /usr/local/bin directory at all. I just reran it and pip put it there this time. I will close this. Thanks for the quick response!

closed 2020-02-14 by BrennaBelletti