zephyrproject-rtos/west #396: Consider using Python type hints

gmarull, 2020-04-24:
Python type hints could be added to the codebase. [mypy](https://mypy.readthedocs.io/en/latest/) static analyser can be used then to check for type errors which otherwise would only be detected at runtime.

mbolivar-nordic, 2020-04-24:
I'm open to the idea. I've followed the type annotation developments since the 2.x days from a distance, but never used them in anger, so there'd be a learning curve for me.

Are you familiar with them? Would you be willing to provide review feedback on the changes?

gmarull, 2020-04-24:
Yes, I've used them on some relatively large Python codebases. I personally find type hints useful and I use them in most new code I write, specially libraries. However, some people do not like to use types on scripting languages. Development experience is also improved on editors supporting type annotations (e.g. VS Code).

I'm open to review or contribute if needed.

mbolivar-nordic, 2020-04-24:
Awesome, thanks for the offer.

Given your experience, would you be willing to start this process off by adding types to a simple API module, like src/west/log.py, and maybe even adding a couple of test cases for type checks that run when `tox` is invoked? That would really help me understand what is needed and how to take it forward from there.

mbolivar-nordic, 2020-06-25:
I think #397 and #405 are a good enough start to close this issue. We can add more type checking to individual API modules as time goes on / as needed. @gmarull if you agree, would you please close?

closed 2020-06-25 by gmarull