zephyrproject-rtos/west #399: West - init/update module SHA with --depth = 1

koffes, 2020-05-11:
When running test build/CI, you are commonly interested in the snapshot at the current SHA, and not the history behind it.

When a project fetches Zephyr or any other module given by a SHA: When the build system runs, this SHA is fetched and along it a large history which is not of interest.
This causes unnecessary traffic and slows down the build.

Today, west supports clone-depth but only for branches and tags.

The desired feature is a clone-depth argument when SHA is used as revision argument.
Even better, west can take an additional argument such as --ci or --build_sys which enforces --depth = 1 on all modules.

This should work as long as the git installation is >= 2.5.0 on both the client and server.
@mbolivar-nordic FYI: @thst-nordic

koffes, 2020-05-11:
This issue is similar to #319

marc-hb, 2020-10-22:
> This issue is similar to #319

Indeed... how different is it?

PS: are you sure your usual git server(s) let you fetch a SHA1 directly? I think it's rare.

koffes, 2020-10-22:
Reading both cases again, I agree that this can be closed and we can continue the discussion in #319.

closed 2020-10-22 by koffes