zephyrproject-rtos/west #400: Add "optional" keyword for given project in manifest

koffes, 2020-05-11:
Sometimes we have support projects for internal testing and internal resources in the manifest file. It can look like this:
```
projects:
    - name: proj1
      path: extra/project-1
      revision: master
    - name: proj2
      path: internal_test/test
      remote: secret_internal_repo
```
When customers get code they also get the manifest file. When they attempt a` west init` or `west update`, they will get an error since the internal_test repo is not accessible outside company firewalls.

We would like to se an "optional" argument for two reasons:
1: Give the customer a clear indication that this repo is not needed to build or use the application.
2: West can give a softer warning when this repo is not accessible. E.g.: West was unable to access secret_internal_repo. However, this project has been labeled optional.

mbolivar-nordic, 2020-05-29:
Hi there, thanks for this. We've discussed the idea of optional projects in the past in other contexts, and my general feeling is I would prefer to implement a slightly more general version of this feature, where repositories can be added to groups, and some groups have special names which are not updated by default. You could then implement optional repositories as a special case.

This would also allow other use cases, like vendor-specific binary blob repositories, that could not be implemented due to IP restrictions in other ways.

Would something like this work for you?

koffes, 2020-06-02:
Thanks @mbolivar-nordic . It sounds like a good idea to have a more general approach.
I believe that could work well, with a minor tweak. You state "not updated by default". For our case I believe West should attempt to update by default, but give a soft warning if the repo is unavailable.

What about creating tags/groups such as:
```
optional_manual #Like you stated, optional repo wich needs to be manually invoked: west update <X>
optional_automatic # Like I stated, will attempt update on `west update` cmd, but fail softly
```

mbolivar-nordic, 2021-04-20:
Repository groups are now part of west. However, I am hesitant to implement the 'optional' bit without a clear use case for why it is necessary.

koffes, 2021-04-21:
Great. I have read the documentation, and it seems that the group feature will provide a very decent solution to this issue. At least it reduces the priority. Let me run a case by you to see if you have any input.

A customer is given a project with the given manifest:
```
manifest:
  projects:
    - name: open_code
      groups:
        - group_open
    - name: closed_code
      groups:
        - group_closed

  group-filter: [-group_closed]
```

This project is on Github, where a limited number of users have access to `closed_code` where as `open_code` is in a open repo.
If the manifest looks like this, it would work well for open users, but closed users will either have to change the group filter in the manifest or run `west config manifest.group-filter +group_closed`. This would work, but requires and extra manual step.
On the other hand, having `group-filter: [+group_closed]` in the manifest would of course not be ideal as it would simply fail for open users and require them to do manual steps.

A possible solution is to add a third option to the groups besides "+" and "-". I.e. ":" which would enable the group by default. -But, if then the repo `closed_code` is not found, it would print: `closed_code not found. You may not have access. This group is marked as optional in the manifest`.

mbolivar-nordic, 2021-04-21:
The expectation is indeed that the extra step would be necessary before using `closed_code`. (CI and other places that just need one-shot workspace setup can use `west update --gf=+group_closed` for brevity.)

I personally don't find the extra step too onerous, and I'm wary of the extra complexity that comes with adding a third state for projects. There were already unexpected consequences just adding a simple binary active/inactive:

https://github.com/zephyrproject-rtos/west/issues/491

I also got the interaction between imports and group filter wrong in v0.9.0 and had to break compatibility in v0.10.0 as mentioned here:

https://docs.zephyrproject.org/latest/guides/west/release-notes.html#v0-10-0

I'm therefore going to lean on the ["conservative about features"](https://docs.zephyrproject.org/latest/guides/west/why.html#design-constraints) design constraint for west, and say that I don't think an explicit "optional" state is a good idea for now. I understand how that one use case could be made more convenient by that, but I don't think it justifies the extra complexity by itself.

marc-hb, 2021-04-21:
> west config manifest.group-filter +group_closed

You could make that command part of your build process. Pseudo build code:
```
 if closed_feature enabled # you should already have line(s) like this one
     if not exists closed_project_clone
          west config manifest.group-filter +group_closed
          west update closed_projects
```

You could also branch the manifest.

Finally, you may be interested in the "import" feature to hide your hidden projects better :-)

mbolivar-nordic, 2021-04-21:
> You could make that command part of your build process. Pseudo build code:

I think this is more about the README for users that might go either way

koffes, 2021-04-22:
I agree with the above comments and the new functionality solves this well. Closing.

closed 2021-04-22 by koffes