zephyrproject-rtos/west #413: Setting remote url-base that requires credentials(private repository)

DanielPiflaks, 2020-07-05:
I'm trying to set on my west manifest file a remote url-base, which is a Git link that requires login (private repository).
When running the "west update" command, it shows:
"fetching, need revision MY-REVISION
fatal: unable to access 'MY-URL-BASE-LINK': The requested URL returned error: 403"
It is important to say that when I run "git fetch" from the same CMD(or any other git command that requires authentication ), the authentication works
Is there a way to configure the manifest or the west tool to use my credentials when trying to fetch from the URL? 
I have tried to look for this issue in the docs and in the issues here but found nothing related.
Thanks!

mbolivar-nordic, 2020-07-06:
Thanks for taking the time to file an issue, and sorry you are having trouble.

That is strange; `west update` just runs `git fetch` internally, so any credential store you have set up to make `git fetch` work in that repository should work exactly the same way when west runs git as when you run git manually in the cmd.exe window.

Would you be able to run `west -v update` (i.e. enable verbose mode) and try running the `git fetch` command that fails when rest runs it in your cmd.exe? If the exact same command works when you run it manually but fails when west runs it, we can start to diagnose the environment issues that are causing this.

You'll need to look for something like this in the output:

```
--- PROJECT_NAME: fetching, need revision SOME_REF
west.manifest: running 'git fetch -f --tags -- https://REPOSITORY_URL SOME_REF' in /SOME/DIRECTORY
From https://REPOSITORY_URL
 * branch            SOME_REF     -> FETCH_HEAD
```

Then copy/paste the quoted 'git fetch' command.

DanielPiflaks, 2020-07-07:
@mbolivar-nordic  Thank you very much for your help!
After I ran `west -v update`, I got the Fetch command as you mentioned.
I took this command and ran it manually from the git repository and I found the problem: on my west.yml I configured the remote URL wrong (forgot that the `- name:` inside the repository adds the name to the URL..)
After fixing the URL, everything works as expected.
Thank you!

closed 2020-07-07 by DanielPiflaks