zephyrproject-rtos/west #419: Feature request:  Allow symlinks for west project directory

sslupsky, 2020-08-07:
The documentation for west here https://docs.zephyrproject.org/latest/application/index.html#creating-an-application indicates that building Zephyr or creating an application in a directory with spaces anywhere on the path is not supported.

I have a project in a system directory that has a space in the path.  For instance, the base directory is:

/Users/stevenslupsky/Library/Mobile Documents/com~apple~CloudDocs/source/zephyrproject

I attempted to work around this problem by creating a symlink as follows:

```
lrwxr-xr-x    1 stevenslupsky  staff         65  7 Aug 11:26 iCloudDocs -> /Users/stevenslupsky/Library/Mobile Documents/com~apple~CloudDocs
```

which allows me to use "iCloudDocs" in place of "Library/Mobile Documents/com~apple~CloudDocs" which eliminates the space in the path.  IE, my ZEPHYR_BASE is:

/Users/stevenslupsky/iCloudDocs/source/zephyrproject/zephyr

However, when I run west topdir, west is reporting the path with the space instead of the symlink:
```
/Users/stevenslupsky/Library/Mobile Documents/com~apple~CloudDocs/source/zephyrproject
```

I tried to set an absolute path in the west config file but that did not work.

It would be helpful if west could be configured to use a custom path.

mbolivar-nordic, 2020-08-07:
This is not the solution you are looking for, I am afraid: https://github.com/zephyrproject-rtos/zephyr/blob/master/CMakeLists.txt#L25

sslupsky, 2020-08-07:
I do not see that error.  I get the same error message when I build with or without the symlink.  The output shows the path with the space in it.  So, I am making a bit of an assumption here, I think the build fails because the space causes a problem.  I moved the directory back to a directory without a space and everything builds properly.  I would like to use the "Mobile Documents" directory because that is what Apple uses for their iCloud drive.  So, I would be able to access my local repo on my other devices (desktop, laptop, iPad, phone).  I unfortunately have no choice with regards to the path that Apple used and cannot avoid the space.  I was hoping the symlink would work but somewhere West must resolve the path to an absolute path that replaces the symlink.

mbolivar-nordic, 2020-08-07:
> I do not see that error. 

Sorry for being terse. What I mean is that the zephyr build system doesn't cope well with symlinks *or* spaces, so my guess is that trying to handle your use case with symbolic links will only make problems worse, not better.

Ultimately I think this build system limitation should be revisited and fixed in Zephyr itself.

> somewhere West must resolve the path to an absolute path that replaces the symlink.

In the case of the `west topdir` command, the issue is that `os.getcwd()` returns the physical current working directory instead of the logical one.

```
$ mkdir foo
$ ln -s foo link-to-foo
$ cd link-to-foo
$ python -c 'import os; print(os.getcwd())'
/tmp/foo
```

West finds the topdir by starting at `os.getcwd()`. Other things like `Path('.').resolve()` and `os.path.abspath('.')` will similarly return the physical path. We could potentially try looking in `os.environ['PWD']` but I would be reluctant to make that the default behavior. Perhaps hide it under a config option?

sslupsky, 2020-08-07:
> West finds the topdir by starting at os.getcwd(). Other things like Path('.').resolve() and os.path.abspath('.') will similarly return the physical path. We could potentially try looking in [...]

Yeah, that is what I thought would be the underlying issue.  A config option would be fine with me.  I realize this is a "special case".  Thank you for the feedback.

mbolivar-nordic, 2020-08-13:
@sslupsky do you think you might be able to contribute this? https://github.com/zephyrproject-rtos/west/pull/313 may be a source of inspiration.

sslupsky, 2020-08-13:
@mbolivar I had a quick look and it surely looks relevant.  Do you recall why it wasn't merged?  The PR is labeled "draft" but it is not clear to me why it couldn't be used in its current form?

mbolivar-nordic, 2020-08-17:
It wasn't merged because there are unaddressed review comments.

marc-hb, 2022-09-22:
cc:
- https://github.com/zephyrproject-rtos/zephyr/issues/50284

cfriedt, 2023-04-11:
Hmm... I just ran into a desparate need for this to be fixed too. I'll patch my version of west locally, but in case anyone wants to use my change, it was pretty unintrusive.

In my case, I do not have access to `git` on the build machine and the Zephyr sources are read-only. We are encouraged to use symbolic links to speed up builds.

```shell
ZEPHYR=/mnt/foo/zephyrproject-rtos/zephyr
cd /tmp
rm -Rf zephyrproject
mkdir -p zephyrproject/.west
ln -sf $ZEPHYR zephyrproject/zephyr
cat << EOF > zephyrproject/.west/config
[manifest]
path = zephyr
file = west.yml
EOF
cd zephyrproject/zephyr

west status && echo $?
```

Output without patch
```shell
Traceback (most recent call last):
  File "/home/cfriedt/zephyrproject/.venv/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/cfriedt/zephyrproject/.venv/lib/python3.8/site-packages/west/app/main.py", line 866, in main
    app.run(argv or sys.argv[1:])
  File "/home/cfriedt/zephyrproject/.venv/lib/python3.8/site-packages/west/app/main.py", line 96, in run
    self.config = west.configuration.Configuration(topdir=self.topdir)
  File "/home/cfriedt/zephyrproject/.venv/lib/python3.8/site-packages/west/configuration.py", line 169, in __init__
    self._local = _InternalCF.from_path(self._local_path)
  File "/home/cfriedt/zephyrproject/.venv/lib/python3.8/site-packages/west/configuration.py", line 69, in from_path
    return _InternalCF(path) if path and path.exists() else None
  File "/home/cfriedt/zephyrproject/.venv/lib/python3.8/site-packages/west/configuration.py", line 76, in __init__
    raise FileNotFoundError(path)
FileNotFoundError: /mnt/foo/zephyrproject-rtos/zephyr/.west/config
```

Output with patch
```shell
west status && echo $?
0
```

```patch
diff --git a/src/west/util.py b/src/west/util.py
index fd577d3..ec0711d 100644
--- a/src/west/util.py
+++ b/src/west/util.py
@@ -8,6 +8,7 @@
 import os
 import pathlib
 import shlex
+import sys
 import textwrap
 from typing import List, Optional, Union
 
@@ -66,7 +67,8 @@ def west_topdir(start: Optional[PathType] = None,
     Like west_dir(), but returns the path to the parent directory of the .west/
     directory instead, where project repositories are stored
     '''
-    cur_dir = pathlib.Path(start or os.getcwd())
+    cwd = os.getcwd() if sys.platform == "win32" else os.environ["PWD"]
+    cur_dir = pathlib.Path(start or cwd)
 
     while True:
         if (cur_dir / '.west').is_dir():
```