zephyrproject-rtos/west #424: pip3 install west error

AbramQuerashi, 2020-08-14:
Not able to install pip3 on WSL2 ubuntu 

$ sudo pip3 install --user -U west

Could not fetch URL https://pypi.org/simple/west/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/west/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
  ERROR: Could not find a version that satisfies the requirement west (from versions: none)
ERROR: No matching distribution found for west
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping

mbolivar-nordic, 2020-08-17:
> Could not fetch URL https://pypi.org/simple/west/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/west/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping

This indicates your system's SSL configuration is not working properly. I don't have any advice as I don't use WSL2, but perhaps it's an antivirus issue? https://stackoverflow.com/questions/55990369/i-cant-use-pip-on-wsl-ubuntu

Can you run `python3 -c 'import ssl'` in your shell and copy/paste the exact results?

It should print no output. If it prints an ImportError or ModuleNotFoundError, it's an issue with how you've installed pip on WSL2.

AbramQuerashi, 2020-08-20:
$ python3 -c 'import ssl'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.8/ssl.py", line 98, in <module>
    import _ssl             # if we can't import it, let the error propagate
ModuleNotFoundError: No module named '_ssl'

mbolivar-nordic, 2020-08-20:
Thanks. From your output, this isn't a west issue or a pip issue, but an issue with the way pip is run within the WSL2 environment, so I'm closing this. You will need to set up a working development environment to proceed.

Note that WSL is not a supported environment for Zephyr development as noted in the getting started guide.

I suggest you look into this based on how you set up Python in WSL2.

closed 2020-08-20 by mbolivar-nordic