zephyrproject-rtos/west #437: Manifest._load_projects() uses escapes_directory() in ways that break zephyr CI

mbolivar-nordic, 2020-09-15:
**Background**

- west generally assumes full control of the workspace and makes no promises about what happens if you use symlinks inside
- for west 0.8, we added a `path-prefix:` key to an `import:` map, allowing projects to be placed in subdirectories
- to make this safe from directory traversal attacks that attempt to overwrite files outside the workspace with attacker chosen content on `west update`, we use `west.util.escapes_directory()` to force a project's path to live inside the workspace and error out if it doesn't

**Issue details**

Let's say I make some project directory `~/workspace/myproject` a symlink to something outside the workspace, like so:

```bash
rm -rf /my/workspace/myproject
ln -s /project/cache/myproject /my/workspace/myproject
west update
```

**Actual behavior**

I will get something like this:

```
FATAL ERROR: Malformed manifest file: /my/workspace/west.yml
  Schema file: .../west/manifest-schema.yml
Hint: project myproject absolute path /my/workspace/myproject is not a subdirectory of topdir /my/workspace
```

This is because internally we use west.util.escapes_directory(), which compares paths using `Path.resolve()`, which eliminates symlinks.

Additionally, Zephyr's CI in `.buildkite/run-ci.sh` is doing something very similar to the toy example in the issue details above: https://github.com/zephyrproject-rtos/zephyr/blob/c6cf8d16736798f1209dbc55cf30dd72d205fa34/.buildkite/run.sh#L39

This is blowing up testing with west 0.8 due to this fatal error: https://github.com/zephyrproject-rtos/zephyr/pull/28373

**Expected behavior 1**

This is, at the very least, misleading because `/my/workspace/myproject` is definitely a subdirectory of `/my/workspace`, so at least we should fix the error message to say something that makes more sense, maybe like this:

```
FATAL ERROR: Malformed manifest file: /home/mbolivar/zp/zephyr/west.yml 
  Schema file: /home/mbolivar/src/west/src/west/manifest-schema.yml
  Hint: project net-tools has real path /tmp/foo, which is is not a subdirectory of topdir /home/mbolivar/zp
```

**Expected behavior 2**

We need to get west 0.8 out the door and keep Zephyr's CI working. I see these different options:

1. try to make this check work with absolute paths in a way that does not resolve symlinks
2. add a new `west update --no-workspace-escape-check` option that keeps the check in by default, but turns it off when requested, and adds this to Zephyr's CI 

I favor option 2 because I think forcing people to jump through a hoop here and there when they are using west in unexpected ways and prioritizing the integrity of the rest of the file system makes the most sense.

This feels spiritually similar to the way git has `core.protectHFS` and `core.protectNTFS` config options that prevent you from creating certain types of repositories by default that do potentially dangerous things.

Note that I am explicitly saying a new *command line option* for `west update` and not a new configuration option for west, because I want to get this done quickly and avoid bikeshedding the name of the configuration option.

mbolivar-nordic, 2020-09-15:
@tejlmand please let me know if the proposed changes seem OK to you.

tejlmand, 2020-09-15:
> Note that I am explicitly saying a new command line option for west update and not a new configuration option for west, because I want to get this done quickly and avoid bikeshedding the name of the configuration option.

I actually think a `west config` option here would be nicer, cause then you could set up this in CI as system wide, and users has the option to only allow it per workspace, if they prefer.

I think `west config workspace.escape-check [true|false]` looks ok.

But will also approve a PR based on a `--no-workspace-escape-check` proposal

mbolivar-nordic, 2020-09-15:
> But will also approve a PR based on a `--no-workspace-escape-check` proposal

Thanks. The reason I want to avoid a new config option is because we already have several different sections in our [built-in options](https://docs.zephyrproject.org/latest/guides/west/config.html#built-in-configuration-options), which are currently:

```
color.ui
manifest.path
update.fetch
zephyr.base
zephyr.base-prefer
```

In other words, we have `color`, `manifest`, `update`, and `zephyr` sections.

Additionally, since west is extensible, we have various config options like `build.dir-fmt` which are supported by Zephyr extensions.

Since it's a shared namespace there is the potential for namespace conflicts, and I want to start picking out a config namespace that is reserved for west itself and shouldn't be touched by extension commands. That should include things like `manifest`, but maybe `update` was a bad idea? Maybe we need to reserve `core.foo` for west itself, and let everyone go nuts outside of it? Etc etc etc.

I don't want to have to decide this before 0.8 gets released, though ;), and would like to avoid adding new core configuration options before having this discussion.

tejlmand, 2020-09-15:
> I don't want to have to decide this before 0.8 gets released

make sense

mbolivar-nordic, 2020-09-15:
As discussed offline, option 2 is not enough, because the subsequent `west list` done by zephyr_module.py will also fail for the same reason. We need another way. It might be easier just to make Zephyr CI respect west's model of the workspace.

mbolivar-nordic, 2020-09-23:
Sorry, I'm just getting to this. Are you saying #439 fixes CI as-is, or is https://github.com/zephyrproject-rtos/zephyr/pull/28515 also required? If we have to patch CI, I'm not sure we have a fix. Just trying to understand.

tejlmand, 2020-09-24:
> Sorry, I'm just getting to this. 

No worries.

> Are you saying #439 fixes CI as-is, or is [zephyrproject-rtos/zephyr#28515](https://github.com/zephyrproject-rtos/zephyr/pull/28515) also required?

Yes, [zephyrproject-rtos/zephyr#28515](https://github.com/zephyrproject-rtos/zephyr/pull/28515) is also required.

> If we have to patch CI, I'm not sure we have a fix. Just trying to understand.

The [zephyrproject-rtos/zephyr#28515](https://github.com/zephyrproject-rtos/zephyr/pull/28515) patch is backward compatible, so that running CI will work also with older `west` version. (which is why CI is green)
What it does, is to create a symlink per project, which works with both old `west` and `0.8`+#439 instead of a symlink on higher level.

Old `west` didn't care for workspace escape, so symlinks could be used anywhere, including project itself.

mbolivar-nordic, 2020-09-24:
Fixed in https://github.com/zephyrproject-rtos/west/commit/4209bacf39a3b242e1df6d7c67989397836552c4

closed 2020-09-24 by mbolivar-nordic