zephyrproject-rtos/west #443: Does not respect python virtualenv

benjizhai, 2020-09-28:
When I run `west build` from virtualenv:

`[1/125] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
FAILED: zephyr/include/generated/kobj-types-enum.h zephyr/include/generated/otype-to-str.h
cd /home/benji/proj/zephyr/zephyr/build/zephyr && **/usr/bin/python3.8** /home/benji/proj/zephyr/zephyr/scripts/gen_kobject_list.py --kobj-types-output /home/benji/proj/zephyr/zephyr/build/zephyr/include/generated/kobj-types-enum.h --kobj-otype-output /home/benji/proj/zephyr/zephyr/build/zephyr/include/generated/otype-to-str.h --kobj-size-output /home/benji/proj/zephyr/zephyr/build/zephyr/include/generated/otype-to-size.h --include /home/benji/proj/zephyr/zephyr/build/zephyr/misc/generated/struct_tags.json
Traceback (most recent call last):
  File "/home/benji/proj/zephyr/zephyr/scripts/gen_kobject_list.py", line 62, in <module>
    import elftools
ModuleNotFoundError: No module named 'elftools'`

it still detects system python instead of the virtualenv one

closed 2020-09-28 by benjizhai

mbolivar-nordic, 2020-09-28:
Hi @benjizhai, thanks for closing this issue. I assume that means it's working for you. I use west in virtualenv every day so I think it's working well for this.

JanneKiiskila, 2022-10-14:
I am having this issue as well.
@mbolivar-nordic  - without activating the virtualenv, can you do pip list?
- If there are modules like cbor, cbor2 and intelhex there - can you remove them and then retry the build?
-

kaessert, 2024-11-27:
The trick is to install west in the virtualenv. In my case i had a global path to the binary. When i did `pip install west` in the virtualenv it worked fine eventually.

marc-hb, 2024-11-28:
Thanks @kaessert, this is what happens if you follow the Getting Started Guide steps _in order_:

https://docs.zephyrproject.org/4.0.0/develop/getting_started/index.html#get-zephyr-and-install-python-dependencies

kaessert, 2024-11-28:
Welcome :) How you mean? If you run them exactly in order it should work fine. The error happens when you're in a virtualenv but running west from outside it. (pip list doesn't show west in your virtualenv but you have somehow a west binary in your PATH)

marc-hb, 2024-11-28:
> If you run them exactly in order it should work fine. 

Yes.