zephyrproject-rtos/west #451: How to using west to checkout a revision of zephyr

nqduy35, 2020-11-20:
Hello, I follow the tutorial of zephyr and download zephyr's newest version. I want to checkout to the tag zephyr-v2.3.0. 
I've read the tutorial of West, but I don't find how to do it. Did I miss something? Do you have any suggestions?

nqduy35, 2020-11-20:
I found the solution: West hasn't supported to it yet but we can use git inside the zephyr folder. 
Suppose that we are in zephyrproject's directory
$ cd zephyr
$ git checkout <VERSION>

closed 2020-11-20 by nqduy35

mbolivar-nordic, 2020-11-23:
> West hasn't supported to it yet but we can use Git inside the zephyr folder.

Hi, this is by design. West does not change the manifest repository contents. The user is responsible for doing this using git.