zephyrproject-rtos/west #460: add support for github (or other CI) authentication tokens

paulquevedojdrf, 2020-12-07:
It would really nice to have an easy way to provide west with credentials so it can fetch private repositories when building from a CI. 

In the case of github, the token used to access a private repo is not always ephemeral so having it potentially sticking around on the build runner after the job completes is kinda worrisome (i.e. if I used `git config --global credential.helper store`)

For the case of a github workflow something like this could work?
```
build.yml

jobs:
  buildStuff:
     steps:
         - name: Checkout repo containing west manifest 
           uses: actions/checkout@v2
         - name: West Update
           run: west update --credentials=$user:$pass
           env:
                user: ${{ secrets.PRIVATE_GITHUB_TOKEN }}
                pass: ''
```

chrta, 2020-12-08:
I just setup a github action runner which is running "west update" to fetch some private github repos. Here is what i done:
1. When building the docker container which acts as the runner (based on https://github.com/myoung34/docker-github-actions-runner), i added the following git config to the Dockerfile:
```
# configure credentials helper for token based authentication via https
# on github and ensure that always https is used instead of ssh on github
# to be able to use token based authentication
RUN git config --global credential.helper '!f(){ sleep 1; echo "username=$GITHUB_TOKEN"; echo "password="; };f' && \
    git config --global url."https://github.com/".insteadOf git@github.com:
```

This ensures that instead of ssh and git always https is used for github urls. Also if git asks for username and password, the provided credentials.helper is run which uses the environment variable GITHUB_TOKEN as the username.

2. In the project i store the PErsonal Access Token as a secret named REPOSITORY_ACCESS_TOKEN.

3. In the github action yml, i provide the GITHUB_TOKEN as a environment variable:

```
jobs:
  ...
  steps:
    - name: West update
        env:
          GITHUB_TOKEN: ${{ secrets.REPOSITORY_ACCESS_TOKEN }}
        run: |
          west update
```

mbolivar-nordic, 2020-12-08:
Hi there, as described in the documentation, there is nothing for west to do:

https://docs.zephyrproject.org/latest/guides/west/repo-tool.html#private-repositories

The features you need are already part of git.

If something is missing from that documentation, please reopen. Thanks!

closed 2020-12-08 by mbolivar-nordic