zephyrproject-rtos/west #463: Unable to use nested folders on Windows

mfiumara, 2021-01-08:
Hi I'm seeing a problem with a manifest that's nested, it seems to only happen on Windows environments and not on Linux.

I have a reproduction path:
```
mkdir test-west
cd test-west
west init -m https://github.com/mfiumara/west-test.git --mr master
```

I encounter the following error:
```
C:\Users\Mattia\evalan>mkdir test-west

C:\Users\Mattia\evalan>cd test-west

C:\Users\Mattia\evalan\test-west>west init -m https://github.com/mfiumara/west-test.git --mr master
=== Initializing in C:\Users\Mattia\evalan\test-west
--- Cloning manifest repository from https://github.com/mfiumara/west-test.git, rev. master
Initialized empty Git repository in C:/Users/Mattia/evalan/test-west/.west/manifest-tmp/.git/
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 3 (delta 0), reused 3 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), 498 bytes | 19.00 KiB/s, done.
From https://github.com/mfiumara/west-test
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
3cdaef778b1c58d67f90dff48df3314e05046f58 refs/remotes/origin/master
Already on 'master'
Branch 'master' set up to track remote branch 'master' from 'origin'.
Traceback (most recent call last):
  File "c:\program files\python39\lib\shutil.py", line 806, in move
    os.rename(src, real_dst)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\Mattia\\evalan\\test-west\\.west\\manifest-tmp' -> 'C:\\Users\\Mattia\\evalan\\test-west\\applications\\some-folder'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\program files\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 780, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 337, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest)
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\site-packages\west\commands.py", line 129, in run
    self.do_run(args, unknown)
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\site-packages\west\app\project.py", line 207, in do_run
    topdir = self.bootstrap(args)
  File "C:\Users\Mattia\AppData\Roaming\Python\Python39\site-packages\west\app\project.py", line 302, in bootstrap
    shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
  File "c:\program files\python39\lib\shutil.py", line 818, in move
    rmtree(src)
  File "c:\program files\python39\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "c:\program files\python39\lib\shutil.py", line 613, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "c:\program files\python39\lib\shutil.py", line 613, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "c:\program files\python39\lib\shutil.py", line 613, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "c:\program files\python39\lib\shutil.py", line 618, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "c:\program files\python39\lib\shutil.py", line 616, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\Mattia\\evalan\\test-west\\.west\\manifest-tmp\\.git\\objects\\3c\\daef778b1c58d67f90dff48df3314e05046f58'
```

Same thing on Linux:
```
mattia@MFIUMARA-LT:~/west-test # west init -m https://github.com/mfiumara/west-test.git --mr master
=== Initializing in /home/mattia/west-test
--- Cloning manifest repository from https://github.com/mfiumara/west-test.git, rev. master
Initialized empty Git repository in /home/mattia/west-test/.west/manifest-tmp/.git/
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 3 (delta 0), reused 3 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), 498 bytes | 498.00 KiB/s, done.
From https://github.com/mfiumara/west-test
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
3cdaef778b1c58d67f90dff48df3314e05046f58 refs/remotes/origin/master
Branch 'master' set up to track remote branch 'master' from 'origin'.
Already on 'master'
--- setting manifest.path to applications/some-folder
=== Initialized. Now run "west update" inside /home/mattia/west-test.
```

Not sure what's happening or if it's a python installation problem? I asked a colleague to try and he encountered the same problem also on Windows it's probably west windows issue.

carlescufi, 2021-01-08:
I can reproduce this, but I think it's a result of placing the project `azure-sdk-for-c` inside the manifest repository, which is not controlled by west. So having a project inside the `self` path is probably what is causing this failure.

mfiumara, 2021-01-08:
@carlescufi Ah so that's the issue. Then it's just a coincidence that this works on Linux?

marc-hb, 2021-01-08:
> since path needs to be a single-level folder right now.

This should be caught and reported better.

> Then it's just a coincidence that this works on Linux?

Maybe shutil.py does the equivalent of `mkdir -p` on Linux.

mbolivar-nordic, 2021-01-13:
Should be fixed in https://github.com/zephyrproject-rtos/west/pull/464.

mbolivar-nordic, 2021-01-13:
> This should be caught and reported better.

I don't think so. It's an unexpected error, so a full stack trace is beautiful and exactly what I needed :).

commit 197bda1c (2021-01-21): west init: fix "self: path: foo/bar"

If the manifest repository path is placed in a nested subdirectory of
the workspace, shutil.move() does not guarantee success. Strangely,
this works on Linux, even though it probably shouldn't, but it fails
on Windows. Fix that and add a regression test.

Fixes: #463
closed 2021-01-21 by mbolivar-nordic