zephyrproject-rtos/west #465: west build  might destroy your repository

tejlmand, 2021-01-15:
If a user accidentally runs `cmake -DBOARD=<board> path/to/sample` at the root of a repository, several files will be created. 
The user of course has the responsibility cleanup this mess.

However, in case the user calls `west build` after such a mistake, `west` will destroy the users complete repository.
And this must not happen.

To reproduce: (***WARNING*** This procedure will wipe your repo.)
```
$ west init -m https://github.com/nrfconnect/sdk-nrf
$ west update
$ cd nrf
$cmake -GNinja -DBOARD=nrf52840dk_nrf52840 samples/bluetooth/peripheral_bms/
# Failed, user doesn't clean up

# 2 months later, now using west
$ west build -b nrf52840dk_nrf52840 samples/bluetooth/peripheral_lbs/
-- west build: making build dir /tmp/ncs/nrf pristine   # <--- west doing a pristine at my toplevel, not in a build folder :-(
-- west build: generating a build system
CMake Error: The source directory "/tmp/ncs/nrf/samples/bluetooth/peripheral_lbs" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/ncs/nrf -S/tmp/ncs/nrf/samples/bluetooth/peripheral_lbs -GNinja -DBOARD=nrf52840dk_nrf52840
# Wow, did this really happen to me !#$@#%#^%$%

$ ls
# Nothing

$ git status
fatal: not a git repository (or any of the parent directories): .git
``` 

and the repo is gone, and ALL local work has been lost.

carlescufi, 2021-01-15:
We should move this to Zephyr, since it's part of `west build`, which is implemented in the main zephyr tree.

tejlmand, 2021-01-15:
Moved here: https://github.com/zephyrproject-rtos/zephyr/issues/31358

closed 2021-01-15 by tejlmand