zephyrproject-rtos/west #473: Using West for generic, non-zephyr projects

ark-, 2021-02-01:
Is there a way to use this tool for managing a generic project?

I like the manifest file format. The idea of being able to just `west update` to pull in dependencies from multiple sources for a project could be quite useful to resolve dependencies.

If west isn't suitable but there are other tools which do this I am all ears.

ark-, 2021-02-01:
Ok so I think I've figured it out from digging through the docs.

Directory structure:

``` 
my-project
    libraries
        west.yml
    <rest of my project files>
```

Where `west.yml` is something like:

``` yaml
remotes:
    - name: my_scm
      url-base: <url here>

defaults:
    remote: my_scm
projects:
    - name: first-dependency
      repo-path: dependency-name
      revision: 1.0.1
      path: libraries/dependency-name

```

Go into root directory:

``` cmd
west init -l libraries
```

mbolivar-nordic, 2021-02-01:
I use west for non-zephyr projects as well. The only issue is that it will complain at you about a missing zephyr project. You can run `west config zephyr.base go-away-west` (or whatever you want to set it to) to silence the warning.

mbolivar-nordic, 2021-02-01:
I'm going to close this as a duplicate of https://github.com/zephyrproject-rtos/west/issues/246 since that issue has more context. Thanks for trying out west!

closed 2021-02-01 by mbolivar-nordic