zephyrproject-rtos/west #477: `west update --gf +<group>` does not add the group to .west/config

RobertGalatNordic, 2021-02-09:
West version: v0.9.0

Reproduce Step:
- Initialize project with group limited repository `west init <url> --mr <tag/branch/commit>`
- Update the project adding the group filter `west update --gf +<group>`

Expected results:

- In file `.west/config` there is an entry `group-filter = +<group>` under section `[manifest]`

Actual result:
- File `.west/config` does not contain `group-filter` under `[manifest]` section.

Additional symptoms of the issue:
When there is no group filter in the config, an Application using the repository under the filter does not build.

Temporary workaround the issue:
The filter can be added to the config manually with `west config manifest.group-filter +<group>`

marc-hb, 2021-02-09:
https://github.com/zephyrproject-rtos/zephyr/pull/31241/files

> It now  supports a ``--group-filter`` option which allows for ONE-TIME modifications  to the set of enabled and disabled project groups.

Command-line option = one-time modifications is consistent with most other command line options I know.

mbolivar-nordic, 2021-02-09:
Hello and thanks for trying the new west features!

However, the expected behavior is not correct. The --group-filter option operates as if the option were appended to the config file. It does not actually change the config file.

Please see the examples for how to change the configuration file: https://docs.zephyrproject.org/latest/guides/west/manifest.html#west-project-group-examples

closed 2021-02-09 by mbolivar-nordic