zephyrproject-rtos/west #478: Support use of `git worktree` in west

tejlmand, 2021-02-10:
Today, the manifest in Zephyr might contain multiple entries of the same repository, but in different versions (SHAs), such as:
- mcuboot for both Zephyr and TF-M
  https://github.com/zephyrproject-rtos/zephyr/blob/master/west.yml#L96-L98
  ```
    - name: mcuboot
      revision: 3fc59410b633a6d83bbb534e43aac43160f9bd32
      path: bootloader/mcuboot
  ```
  https://github.com/zephyrproject-rtos/zephyr/blob/master/west.yml#L147-L150
  ```
    - name: tfm-mcuboot # This is used by the trusted-firmware-m module.
      repo-path: mcuboot
      path: modules/tee/tfm-mcuboot
      revision: 1.7.0-rc1
  ```

- mbedtls in Zephyr and TF-M (mbedtls is cloned during build time (ninja), but there is work ongoing to see if could be improved):
  https://github.com/zephyrproject-rtos/zephyr/blob/master/west.yml#L93-L95 
   ```
    - name: mbedtls
      revision: 24d84ecff195fb15c889d9046e44e4804d626c67
      path: modules/crypto/mbedtls
  ```
  https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/96340fb6c0b7e31c2070e1f428ca24076d2700a6/trusted-firmware-m/lib/ext/mbedcrypto/CMakeLists.txt#L22-L23
  ```
            GIT_REPOSITORY https://github.com/ARMmbed/mbedtls.git
            GIT_TAG ${MBEDCRYPTO_VERSION}
  ```

This means that `west` will clone and update the same repository multiple times (mcuboot) instead of a single fetch.

If `git workspace` was used, then a single fetch would be enough, and the two paths would be sharing same repo but having different versions.

As example, we could have:
```
    - name: mcuboot
      revision: 3fc59410b633a6d83bbb534e43aac43160f9bd32
      path: bootloader/mcuboot
      worktrees:
        - name: tfm-mcuboot # This is used by the trusted-firmware-m module.
          path: modules/tee/tfm-mcuboot
          revision: 1.7.0-rc1
```
would allow to just fetch mcuboot once, but still allow for two different versions to be used for Zephyr and TF-M.

Similar improvement will be possible for mbedtls in future.

oyvindronningstad, 2021-02-10:
Can this be auto-detected by seeing that another entry has the same URL? That would make this work across separate manifests.

tejlmand, 2021-02-10:
> Can this be auto-detected by seeing that another entry has the same URL? That would make this work across separate manifests.

Good question. I could fear potential issues for repos that is both used upstream and downstream, as to which repo should be the main repo, and which should be the worktree.

One risk I see is if user deletes the main repo, cause he believe it's not needed anymore, as example deleting `modules/tee/tfm-mcuboot` cause he no longer has this project in his downstream manifest.
But `west` had chosen to use that project as the main git repo, and `bootloader/mcuboot` as the worktree.

mbolivar-nordic, 2021-02-10:
For historical context, the common `west --zephyr-base` argument was actually an attempt to try to support worktree, but it hasn't proven useful in practice.

However, I'm not sure this is a use case for `git worktree`. I think worktrees, like local branches other than manifest-rev, should be in the control of the user. We start to run into collisions if west uses it too.

marc-hb, 2021-02-10:
Speaking of `manifest-rev`: it would have to be come an array, one element per project.

> as to which repo should be the main repo, and which should be the worktree.

BTW here's what git-repo does or at least used to do:
```
src/third_party/bluez/.git/
├── config -> ../../../../.repo/projects/src/third_party/bluez.git/config
├── description -> ../../../../.repo/project-objects/chromiumos/third_party/bluez.git/description
├── HEAD
├── hooks -> ../../../../.repo/project-objects/chromiumos/third_party/bluez.git/hooks
├── index
├── info -> ../../../../.repo/project-objects/chromiumos/third_party/bluez.git/info
├── logs -> ../../../../.repo/projects/src/third_party/bluez.git/logs
├── objects -> ../../../../.repo/project-objects/chromiumos/third_party/bluez.git/objects
├── packed-refs -> ../../../../.repo/projects/src/third_party/bluez.git/packed-refs
├── refs -> ../../../../.repo/projects/src/third_party/bluez.git/refs
├── rr-cache -> ../../../../.repo/project-objects/chromiumos/third_party/bluez.git/rr-cache
├── shallow -> ../../../../.repo/projects/src/third_party/bluez.git/shallow
└── svn -> ../../../../.repo/project-objects/chromiumos/third_party/bluez.git/svn

```

tejlmand, 2021-02-15:
> However, I'm not sure this is a use case for `git worktree`. I think worktrees, like local branches other than manifest-rev, should be in the control of the user.

Well, for a start, a worktree is identified by its path:
https://git-scm.com/docs/git-worktree#Documentation/git-worktree.txt-ltworktreegt
> `<worktree>`
>
>   Working trees can be identified by path, either relative or absolute.

and in this case, the path is the path of the project,
In this example, where I use a worktree for `tfm-mcuboot`:
```
git worktree list
/projects/github/ncs/bootloader/mcuboot       858dd03 (detached HEAD)
/projects/github/ncs/modules/tee/tfm-mcuboot  81d19f0 (detached HEAD)
```
I can't see how this limits or conflicts with how the user may use worktrees today.

A users can still create a worktree in any other location they like.
Of course they cannot choose a path from the `manifest` file, but that is true already today, where that path contains a complete git repo.

> Speaking of manifest-rev: it would have to be come an array, one element per project.

@marc-hb  That is correct cause all branches are shared between the repo and the worktree, which means we cannot use the same `manifest-rev` revision in both places, as they most likely will need to point to different SHAs.

> BTW here's what git-repo does or at least used to do:

In `git` itself, the `.git` inside the worktree is just a plain txt file with info of main repo:
```
$ cat /projects/github/ncs/modules/tee/tfm-mcuboot/.git
gitdir: /projects/github/ncs/bootloader/mcuboot/.git/worktrees/tfm-mcuboot
```

mbolivar-nordic, 2021-02-19:
> I can't see how this limits or conflicts with how the user may use worktrees today.

The same page says:

> The new working directory is linked to the current repository, sharing everything except working directory specific files such as HEAD, index, etc.

And it concludes:

> BUGS
> Multiple checkout in general is still experimental

None of this seems conflict-free at all; it actually all sounds pretty scary.

I also see problems related to the user running `west update` if the manifest moves things related to this:

> If you manually move a linked working tree, you need to update the gitdir file in the entry’s directory.

marc-hb, 2021-02-19:
> That is correct cause all branches are shared between the repo and the worktree,

This would be a major interface change BTW. I think I would like it but not everyone might.

I think the asymmetry between "main" worktree and "linked" worktrees is unacceptable for "naive" users who don't know and don't want to know about worktrees. If I (re)move a project right now for testing, cleaning or any other reason, then there is no difference between (re)moving `mcuboot` or `tfm-mcuboot`. It would be extremely confusing if there were a difference. (plus west still works which is also great).

Avoiding such confusing asymmetry is probably the main reason why grepo maintains centralized bare clones in a private location and why "every project is a linked project": https://gerrit.googlesource.com/git-repo/+/refs/heads/main/docs/internal-fs-layout.md#project-objects These also provide obvious caching/performance benefits.

Note grepo recently got a new `--worktree` option that attempts to replace its older, pre-worktree solution with "native" git worktree: https://bugs.chromium.org/p/gerrit/issues/detail?id=11486 The same centralized bare clones approach is used in both implementations.

There are other interesting design considerations and caveats at both references, recommended.

> If you manually move a linked working tree, you need to update the gitdir file in the entry’s directory.

From https://bugs.chromium.org/p/gerrit/issues/detail?id=11486#c1
> (3) we strongly prefer to have paths be relative.  this allows people to relocate repo client checkouts without all the paths breaking and effectively forcing devs to create a new checkout from scratch.  for example, this is supported

cc: @vapier

vapier, 2021-02-20:
the default [git-repo](https://gerrit.googlesource.com/git-repo) behavior of using symlinks for internal .git state to share objects has a number of downsides that can lead to data loss if you try to use it for more than one checkout.  i would not recommend it at all.  `git gc` and related operations run in the leaf repo and don't see any of the others, so it can discard objects that the other repos are using.  and the worst part is that these operations tend to kick in weeks/months after use :(.  i prob should add a warning note block to the internal-fs-layout.md file in case someone tries to use it as a model for their own thing.

worktrees avoid that because every leaf repo has visibility into refs of all other repos.  i get that the behavior will be diff and a bit weird at first, but i would strongly recommend trying to utilize standard git features rather than do your own thing behind git's back whenever possible.

along those lines, git itself is pushing a ["superproject" approach](https://en.wikibooks.org/wiki/Git/Submodules_and_Superprojects) for creating monorepos from multiple repositories.  we have people in Google seeing if they can use this flow to replace [git-repo](https://gerrit.googlesource.com/git-repo) usage entirely for Android & Chromium OS, and adding features to git when something is missing.

wrt relative paths, [git-repo](https://gerrit.googlesource.com/git-repo) worksaround it by manually rewriting the .git files.  it's ugly, but seems to be working.  i filed a report for git to hopefully support this itself someday.
https://bugs.chromium.org/p/git/issues/detail?id=47

> Can this be auto-detected by seeing that another entry has the same URL? That would make this work across separate manifests.

fwiw, in [git-repo](https://gerrit.googlesource.com/git-repo) we specifically avoid using the base URIs as indexes to support mirror switching.  [we allude to that in the doc Marc referenced earlier](https://gerrit.googlesource.com/git-repo/+/refs/heads/main/docs/internal-fs-layout.md#project-objects):
> This layout is designed to allow people to sync against different remotes (e.g. a local mirror & a public review server) while avoiding duplicating the content. However, this can run into problems if different remotes use the same path on their respective servers. Best to avoid that.

mbolivar-nordic, 2021-02-23:
> the default [git-repo](https://gerrit.googlesource.com/git-repo) behavior of using symlinks for internal .git state to share objects has a number of downsides that can lead to data loss if you try to use it for more than one checkout. i would not recommend it at all.

This use of symlinks was a dealbreaker for using git-repo for Zephyr, actually, since Windows is a supported host OS. I was using repo at my previous company for Zephyr development, but we couldn't get that approach upstreamed because of this. That eventually led to us going our own way with west.

> i would strongly recommend trying to utilize standard git features rather than do your own thing behind git's back whenever possible.

Agreed. I don't think what we're currently doing in west is going behind git's back, though -- I don't think that's what you're saying, just pointing this out FTR.

> along those lines, git itself is pushing a ["superproject" approach](https://en.wikibooks.org/wiki/Git/Submodules_and_Superprojects) for creating monorepos from multiple repositories. we have people [...]

That's an interesting long term project.

Zephyr users rely on west's [manifest import](https://docs.zephyrproject.org/latest/guides/west/manifest.html#manifest-imports) feature to pull in git repo definitions from upstream in flexible ways. I'm not aware of any work in submodules which provides similar flexibility.

I'd be curious if you have similar use cases. The design for west's `import` was me scratching my own itch in response to frustrations with repo's [include](https://gerrit.googlesource.com/git-repo/+/master/docs/manifest-format.md#Element-include) feature.

tejlmand, 2021-02-24:
> > I can't see how this limits or conflicts with how the user may use worktrees today.
> 
> The same page says:
> 
> > The new working directory is linked to the current repository, sharing everything except working directory specific files such as HEAD, index, etc.
> 

Which was actually the intention of this idea. Instead of `NxFull repo` size, there would simply be `1xFull repo + (N-1)xHead` size.
Share everything that can be shared, so that it doesn't have to be fetched multiple times.

> And it concludes:
> 
> > BUGS
> > Multiple checkout in general is still experimental
> 

This, I agree, is a no-go.
Although I have never experienced strange behavior when using worktrees, and I find them useful in certain cases, then this reference means we should no go down this path yet.

The last thing users want, is a strange behaving repo tool.
Users want to focus on development, not a repo tool that can behave unpredictably in edge cases.

> I also see problems related to the user running `west update` if the manifest moves things related to this:
> 
> > If you manually move a linked working tree, you need to update the gitdir file in the entry’s directory.

This in particular is not a concern to me, as `west` could be using `git worktree move` in this case.
If a user manually moves or deletes the worktree, it would just be similar to a user moving or deleting a project today.
Next `west update` would find that it is missing, and recreate it.

tejlmand, 2021-02-24:
@vapier thanks for a very interesting comment.

tejlmand, 2021-02-24:
closing, cause as @mbolivar-nordic points out, this is stated regarding `git` worktrees:
https://git-scm.com/docs/git-worktree#_bugs
> **BUGS**
>
> Multiple checkout in general is still experimental, and the support for submodules is incomplete. It is NOT recommended to make multiple checkouts of a superproject.

I don't want to propose `west` to use any feature that is not considered a complete / stable feature.
If this changes in future, this proposal can be re-opened.

closed 2021-02-24 by tejlmand

mbolivar-nordic, 2021-02-24:
> This in particular is not a concern to me, as `west` could be using `git worktree move` in this case.

How? If the manifest moves a project directory, west has no idea where the old one is in the workspace, because `west update` doesn't pull.

mbolivar-nordic, 2021-02-24:
(I know you closed this @tejlmand , but I think the other issue is a dealbreaker too, unless I'm misunderstanding something.)

tejlmand, 2021-02-24:
> > This in particular is not a concern to me, as `west` could be using `git worktree move` in this case.
> 
> How? If the manifest moves a project directory, west has no idea where the old one is in the workspace, because `west update` doesn't pull.

Right, if we strictly mimic today's behavior on how `west` simply abandon a project that is moved / removed in a manifest.
`west` would leave the old project / repo in the workspace and never clean up, and just create a new project at the new location.
That would be no different than to add a new worktree, and rely on the user to remove the abandoned one.

However, with worktrees, `west` would in principle be able to use the `git worktree list` to obtain knowledge of worktrees, and through that issue a move instead of abandoning.
This would of course require a form of understanding, similar to `manifest-rev`, to ensure `west` wouldn't move a user created worktree.

But I guess exploring such options makes little sense now.