zephyrproject-rtos/west #487: Add group filter to more built-in commands

mrrosen, 2021-03-16:
Currently, only `west update` supports the `--group-filter`/`--gf` command line argument, but it would be good if other commands such as `west forall` and `west status` also allowed users to run commands on a subset of projects using groups on the commandline (without having to list out a bunch of projects in the command line like `west status proj1 proj2 proj3 proj4`, using instead `west status --gf=+mygroup`.

mbolivar-nordic, 2021-03-16:
> rather than `west status proj1 proj2 proj3 proj4`, be able to `west status --gf=+mygroup` instead

Sorry, but I'm a bit confused about this use case. Those two commands would only produce the same result if all projects are inactive by default. Are you sure this is what you meant?

mrrosen, 2021-03-16:
@mbolivar-nordic I might be misunderstanding something as I only recently started looking into groups so Ill give you the concrete thing I was trying to do. In our codebase, we have a manifest repository which includes a few internal repos as projects and a few external repos (including zephyr) as projects. Something like this:

    projects:
      - name: internal1
        revision: some-branch
      - name internal2
        revision some-branch
    
      - name: zephyr
        remote: zephyr
        revision: v2.5.0
        import:
          name-whitelist:
          - proj1
          - proj2
      - name: external1
        remote: external-repo
        revision: v1.1.0

Note, there are many more internal and external projects. While all are enabled, I want to do some action (like create a tag) on all of our internal projects repos. I thought adding them all to a group would let me do this using `west forall` but it doesnt have that functionality right now. I dont necessarily want to disable projects (though thats something we might want to do) across multiple commands, just run a command across a subset of our repos without having to explicitly list them all out.

mrrosen, 2021-03-16:
For example, to solve my current problem, I still used `west forall` but used the fact all of our internal projects share the same remote. It was a bit ugly, thus why I was looking into groups. (For example, something like `west forall -c "git ls-remote --exit-code internal-remote && git tag MyTag"`)

mbolivar-nordic, 2021-03-16:
OK, so this isn't as much about the group-filter feature as doing some action on all projects with a particular group or groups.

I would prefer to call that option `-g / --group`. I can add it to the forall command and all the forall-alikes (status, diff, ...).

So you could say `west status --group foo` and that's the same thing as `west status proj1 proj2 ... projN` for projects 1 through N with group `foo`.

Is that what you're asking for?

mrrosen, 2021-03-16:
> OK, so this isn't as much about the group-filter feature as doing some action on all projects with a particular group or groups.
>
> I would prefer to call that option `-g / --group`. I can add it to [...]

Yes, thats exactly what Im looking for

marc-hb, 2021-03-16:
> OK, so this isn't as much about the group-filter feature as doing some action on all projects with a particular group or groups.
> I would prefer to call that option -g / --group. I can add it to the forall command and all the forall-alikes (status, diff, ...).

I don't understand what makes `west update` different and why its group option should be named differently from `west status`, `diff` and all the other commands.

mbolivar-nordic, 2021-03-16:
> I don't understand what makes `west update` different and why its group option should be named differently from `west status`, `diff` and all the other commands.

Because this is not affecting the group filter at all, it's doing something different. See my original comment in the thread

marc-hb, 2021-03-16:
Ah yes of course, I clearly didn't understand the command line `--group-filter` option. I didn't realize it's not standalone but combined with everything else. I suspect @mrrosen rosen made a similar confusion.

So @mrrosen you're indeed asking for a new, different feature that does not exist yet. Please update the title of this issue.

The next questions that come to mind are then: 1. should `west update` get the new feature too? 2. Should other commands get the existing `--group-filter` too? No idea yet.

mrrosen, 2021-03-16:
I guess I am since Im not 100% sure I understand the group-filter feature. Im not sure what a better title is as I still want a "group filter" even though it isnt directly using "group-filter"

marc-hb, 2021-03-16:
Like everything in west, `--group-filter` has excellent and extensive documentation https://docs.zephyrproject.org/latest/guides/west/manifest.html#group-filter

`west help update` also has a useful clue:
```
  --group-filter FILTER, --gf FILTER
     proceed as if FILTER was _appended_ to manifest.group-filter; may be given multiple times
```
Emphasis mine.

The long story short: `--group-filter` can only _alter_, not replace the default selection and most repos are (obviously) selected by default. So to `west status`  a small subset you would typically have to exclude a long list of groups on the command line which does not sound like what you're looking for. Group filters also combine all with each other (command line + config + manifests) with some precedence rules which again does not sound like the simpler feature you're after.

Hence @mbolivar-nordic suggesting a new option with a different name and me suggesting you rephrase the title of this issue.

Hope this helps.